libimobiledevice/libimobiledevice #1452: idevicedebug + ideviceimagemounter broken with iOS 17

jawshoeadan, 2023-06-06:
There seems to be a new structure or system for developer disk images and invoking debugserver. Debugging through Xcode in the "intended" way works the same. 

Firstly, after iOS 16.4, there are no more disk images and signatures in the Xcode directory.
[image]

Second, when first plugging my iOS 17 iPhone into Xcode 15, I saw this message:
[image]

Also, I noticed there are certain specific device disk images at /Applications/Xcode-15-beta.app/Contents/Resources/CoreDeviceDDIs/iOS_DDI.dmg

[version.plist.txt](https://github.com/libimobiledevice/libimobiledevice/files/11659084/version.plist.txt)
[image]

https://mastodon.social/@siguza/109922466146143055

mexmer, 2023-06-06:
this is not issue with libimobile, apple need to fix this, provide proper developer image for new OS version, then and only then, we can see how libimobile works with that.

Dantee296, 2023-06-07:
seems with ios 17 we need to modify image mounter to support "personalized developer images"

zhuowei, 2023-06-07:
@doronz88 Apple did mention they switched to a new framework for device communication (https://techhub.social/@JTostitos/110499792791802006).
Is the RemoteServiceDiscovery the same RemoteXPC framework used to talk with the Touch Bar / Studio Display (https://duo.com/labs/research/apple-t2-xpc)?

doronz88, 2023-06-08:
@zhuowei it does bare some resemlence, at least the fact its over USB ethernet (ncm). I'm still trying to wrap my head arround it

doronz88, 2023-06-08:
UPDATE: I'm getting the hang of it. It is RemoteXPC, but the existing tools will require some work before they can actually capture this

mexmer, 2023-06-08:
i'm kind of consused, so it doesn't go anymore on top of their mux protocol, but instead over tcp on top of rndis adapter (or wifi)?

still kinda surprised it doesn't contact lockdown first to start mounter service, that would mean, that service is either running all time, or is started from xpc proxy on phone.

doronz88, 2023-06-08:
The mounter service is still on top of `lockdownd` as you can see in my implementation over here:
https://github.com/doronz88/pymobiledevice3/pull/463/files

However, the DeveloperDiskImage interactions are purely over network afterwards. I'm currently working on a sniffer and parser for packets sent from the Interuments app and publish it once it is completed.

karthikgbbristol, 2023-06-08:
That makes sense, @doronz88 I was tying to use socat over usbmux and found no traffic on it when running Instruments, making me wonder if they are using another interface.

doronz88, 2023-06-08:
I'm currently sniffing it from my `en19` network which is created when I connect the device over USB

jawshoeadan, 2023-06-13:
@doronz88 Which files are the trust cache and build manifest? Where do you find each?

doronz88, 2023-06-14:
> @doronz88 Which files are the trust cache and build manifest? Where do you find each?

Inside the `IOS_DDI.dmg`

Geczy, 2023-06-14:
where do you find the signature, im4m?

doronz88, 2023-06-15:
It depends on the device nonce. Meaning it will only work till device reboots

doronz88, 2023-06-15:
I'll add a CLI method, but anyway you can trigger the `get_manifest_from_tss()` and provide the correct build manifest

Geczy, 2023-06-15:
ah got it, yeah i can get the correct build manifest. thanks ill try that for now

Geczy, 2023-06-15:
fwiw, this can all be extracted automatically

```py
@mounter.command('mount-personalized', cls=Command)
@click.argument('image', type=click.Path(exists=True, file_okay=True, dir_okay=False))
@click.argument('trust-cache', type=click.Path(exists=True, file_okay=True, dir_okay=False))
@click.argument('build-manifest', type=click.Path(exists=True, file_okay=True, dir_okay=False))
@catch_errors
def mounter_mount_personalized(lockdown: LockdownClient, image: str, trust_cache: str, build_manifest: str):
    """ mount personalized image """
    PersonalizedImageMounter(lockdown=lockdown).mount(Path(image), Path(build_manifest), Path(trust_cache))
    logger.info('Personalized image mounted successfully')
```

from `/Applications/Xcode-beta.app/Contents/Resources/CoreDeviceDDIs/iOS_DDI.dmg`

why the need to specify it one at a time?

Geczy, 2023-06-15:
```sh
2023-06-15 11:58:41 M7V6KFVVNL pymobiledevice3.cli.mounter[81858] ERROR Given image was already mounted
➜  ~ pymobiledevice3 developer simulate-location set 0 0
Readline features including tab completion have been disabled because
no supported version of readline was found. To resolve this, install
pyreadline3 on Windows or gnureadline on Linux/Mac.

2023-06-15 11:58:46 M7V6KFVVNL pymobiledevice3.__main__[81865] ERROR Failed to access an invalid lockdown service, possibly from DeveloperDiskImage.dmg or a Cryptex. You may try: python3 -m pymobiledevice3 mounter auto-mount
```

huh ?

doronz88, 2023-07-17:
Just updating I already have a part of the Developer Services working.
For the image mounter you can look and use the same implementation for `libimobiledevice`:
https://github.com/doronz88/pymobiledevice3/blob/master/pymobiledevice3/services/mobile_image_mounter.py#L325

In general, I created this repository which stores all the DeveloperDiskImages together with the new Personalized format as splitted files so we'll be able to mount them cross-plartform.
https://github.com/doronz88/DeveloperDiskImage

mforys, 2023-08-07:
When do you expect to have these fixes already merged to the official release branch?

karthikgbbristol, 2023-08-09:
Also, has anyone tested if this  fork is able to retrieve performance metrics like Memory, CPU Usage etc.,

kj54321, 2023-09-22:
Hi Folks, any update for this issue? pymobiledevice3 only has work around on MacOS, but still no luck on Windows/Linux. Thanks

jzhang1043, 2023-11-17:
`(myenv) ➜  ~ python -m pymobiledevice3 mounter auto-mount --rsd fdad:84fd:1e10::1 59338
Traceback (most recent call last):
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/__main__.py", line 125, in <module>
    cli()
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/__main__.py", line 77, in cli
    cli_commands()
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/cli/cli_common.py", line 133, in wrap_callback_calling
    callback(service_provider=service_provider, **kwargs)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/cli/mounter.py", line 122, in mounter_auto_mount
    auto_mount(service_provider, xcode=xcode, version=version)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/services/mobile_image_mounter.py", line 362, in auto_mount
    auto_mount_personalized(lockdown)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/services/mobile_image_mounter.py", line 355, in auto_mount_personalized
    PersonalizedImageMounter(lockdown=lockdown).mount(image, build_manifest, trustcache)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/services/mobile_image_mounter.py", line 199, in mount
    self.raise_if_cannot_mount()
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/services/mobile_image_mounter.py", line 34, in raise_if_cannot_mount
    if Version(self.lockdown.product_version).major >= 16 and not self.lockdown.developer_mode_status:
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/remote/remote_service_discovery.py", line 45, in developer_mode_status
    return self.lockdown.developer_mode_status
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/lockdown.py", line 291, in developer_mode_status
    return self.get_value('com.apple.security.mac.amfi', 'DeveloperModeStatus')
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/lockdown.py", line 87, in _inner_reconnect_on_remote_close
    return f(*args, **kwargs)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/lockdown.py", line 408, in get_value
    res = self._request('GetValue', options)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/lockdown.py", line 515, in _request
    response = self.service.send_recv_plist(message)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/service_connection.py", line 123, in send_recv_plist
    self.send_plist(data, endianity=endianity, fmt=fmt)
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/service_connection.py", line 169, in send_plist
    return self.sendall(build_plist(d, endianity, fmt))
  File "/opt/homebrew/anaconda3/envs/myenv/lib/python3.10/site-packages/pymobiledevice3/service_connection.py", line 118, in sendall
    self.socket.sendall(data)
OSError: [Errno 9] Bad file descriptor
`

why am I having this error message?

jzhang1043, 2023-11-18:
> https://github.com/doronz88/pymobiledevice3/blob/master/pymobiledevice3/services/mobile_image_mounter.py#L325

nvm, I reinstalled python from 3.10 to 3.11 and the problem was resolved.

gotwig, 2023-12-10:
When I understand correctly there is a fix branch that just needs to be merged on main?
Is that possible soon :)?

mexmer, 2023-12-10:
no, libimobile has no support for this. there exists python implementation, if it works on other systems than os x, no idea, either test it, or ask author

https://github.com/doronz88/pymobiledevice3