libimobiledevice/libimobiledevice #1456: ideviceenterrecovery Failed to enter recovery mode. LOCKDOWN_E_SESSION_INACTIVE

MuhammedElsaid, 2023-06-11:
I'm trying to use ideviceenterrecovery and it fails. When I debug the code and see the return value of
`ret = lockdown_check_result(dict, "EnterRecovery")`
it's LOCKDOWN_E_SESSION_INACTIVE.

if I'm using idevicediagnostics 
I'm testing it on iPhone 7 Plus.

Windows 11
Latest version of iTunes

Latest build of libmobiledevice

iOS version 15.7.6
Find My iPhone is active.

Thanks a lot.

nikias, 2023-06-11:
Latest code? It needs a lockdown session so the device has to be paired with your computer.

mexmer, 2023-06-12:
as of iOS 15.1 device need to be unlocked and paired to reboot from commandline (either normal reboot, or put into recovery mode). before you could reboot unpaired or even locked device (if it didn't have usb restriction enabled)

MuhammedElsaid, 2023-06-13:
> as of iOS 15.1 device need to be unlocked and paired to reboot from commandline (either normal reboot, or put into recovery mode). before you could reboot unpaired or even locked device (if it didn't have usb restriction enabled)

It's paired and fully unlocked and it doesn't have any USB restriction.

MuhammedElsaid, 2023-06-13:
> Latest code? It needs a lockdown session so the device has to be paired with your computer.

I was using an old version for debugging the problem. Thank you for pointing it out.
I still get `Failed to enter recovery mode.` when I use the latest build version from msys2 [repo](https://packages.msys2.org/package/mingw-w64-clang-x86_64-libimobiledevice?repo=clang64)

Sam-Neale, 2024-01-17:
Anyone had any luck with this?, am getting an identical thing with iOS 17.3 iPhone 15 Pro

Sam-Neale, 2024-01-17:
[image]
If it's any help, here is the debug for my attempt. Yes the UUID is correct.

Sam-Neale, 2024-01-17:
Nevermind... seems this must have been fixed between v1.3.0 and current head. Hopefully a new version soon?