libimobiledevice/libimobiledevice #1457: idevicebackup2 ERROR: Could not receive from mobilebackup2 (-4)

olly1240, 2023-06-12:
As in the title when I try to start a backup using idevicebackup2 i get the following error 
`ERROR: Could not receive from mobilebackup2 (-4)`
`Received 0 files from device`

When I start a backup my phone asks for the lockscreen code then, after inputting it, it shows the backup icon circling, after a while idevicebackup2 returns the error. 

I tried different cables, I also tried using netmuxd, reinstalled from scratch the whole libimobiledevice stack, but to no avail.

I am on idevicebackup2 1.3.0-179-g860ffb7

olly1240, 2023-06-15:
Fixed by recompiling with OpenSSL instead of GNUtls

besendorf, 2023-06-15:
I had the same issue. What exactly did you do?

olly1240, 2023-06-15:
> I had the same issue. What exactly did you do?

What distro are you on?

olly1240, 2023-06-15:
> Arch

Get libimobiledevice-git from aur then edit the PKGBUILD, replace `--without-openssl` with `--with-openssl` and delete `--with gnutls`