libimobiledevice/libimobiledevice #1459: Can i use it on iphone?

xanadu-lee, 2023-06-15:
Is there some way that I can call this dynamic library on the iPhone and link itself?
For example, find the WIFI connection port supported by the current device in some way?
Or simulate USB data sent to the machine?

Schlaubischlump, 2023-06-15:
If I understand you correctly you are asking if you can use this library inside an iOS app ?

I don't think that's possible, since there is no usbmuxd on iOS. You would first need to find an usbmuxd implementation that runs on iOS.

xanadu-lee, 2023-06-19:
Your understanding can be said to be partially correct.
My thought is that I should be able to connect to this device with the WiFi connection allowed. The usbmuxd is not needed at this time, since it is not forwarded through USB.
But not all devices will enable this feature.
So I'm looking for a way to pretend that the device is connected to the USB on the PC side.
There is actually no need for usbmuxd, because there is no need to enumerate devices at this time, and all communication through usbmuxd must be redirected to the simulated channel.