libimobiledevice/libimobiledevice #1464: if your distribution doesn't contain this package, you need to compile from source code

RolfKausH, 2023-06-26:
if your distribution doesn't contain this package, you need to compile from source code

Can you give me some guidance on how to start the compilation? I'm a newbie

AiXanadu, 2023-06-27:
# install dependencies
sudo apt update
sudo apt upgrade
sudo apt install libtool
sudo apt install make
sudo apt install automake
sudo apt install autoconf
sudo apt install libreadline-dev
sudo apt install python
sudo apt install pkg-config
sudo apt install vim
sudo apt install git

# set environment variables
export XXX_PREFIX=${HOME}/binary/linux-x64
export PKG_CONFIG_PATH=${XXX_PREFIX}/lib/pkgconfig
export CFLAGS=-I${XXX_PREFIX}/include
export LDFLAGS=-L${XXX_PREFIX}/lib
export PATH=${PKG_CONFIG_PATH}:${PATH}

# Compile parameters and order
libplist-2.3.0/autogen.sh --prefix=${XXX_PREFIX} --without-cython PACKAGE_VERSION=2.3.0
libimobiledevice-glue-1.0.0/autogen.sh --prefix=${XXX_PREFIX} --without-cython PACKAGE_VERSION=1.0.0
libusbmuxd-2.0.3/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=2.0.3
libimobiledevice-1.3.1/autogen.sh --prefix=${XXX_PREFIX} --without-cython PACKAGE_VERSION=1.3.1
libirecovery-1.0.1/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=1.0.1
libideviceactivation-1.1.2/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=1.1.2
idevicerestore-1.0.1/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=1.0.1
usbmuxd-1.1.2/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=1.1.2

RolfKausH, 2023-06-27:
Thanks a lot, 怯花颜I will try and get back to you. Am 27.06.23 um 03:10 schrieb 怯花颜
                
            
            
                
                    Von: "怯花颜" ***@***.***>Datum: 27. Juni 2023An: "libimobiledevice/libimobiledevice" ***@***.***>Cc: "RolfKausH" ***@***.***>,"Author" ***@***.***>Betreff: Re: [libimobiledevice/libimobiledevice] if your distribution doesn't contain this package, you need to compile from source code (Issue #1464)
                
                
install dependencies
sudo apt update
sudo apt upgrade
sudo apt install libtool
sudo apt install make
sudo apt install automake
sudo apt install autoconf
sudo apt install libreadline-dev
sudo apt install python
sudo apt install pkg-config
sudo apt install vim
sudo apt install git
set environment variables
export XXX_PREFIX=${HOME}/binary/linux-x64
export PKG_CONFIG_PATH=${XXX_PREFIX}/lib/pkgconfig
export CFLAGS=-I${XXX_PREFIX}/include
export LDFLAGS=-L${XXX_PREFIX}/lib
export PATH=${PKG_CONFIG_PATH}:${PATH}
Compile parameters and order
libplist-2.3.0/autogen.sh --prefix=${XXX_PREFIX} --without-cython PACKAGE_VERSION=2.3.0
libimobiledevice-glue-1.0.0/autogen.sh --prefix=${XXX_PREFIX} --without-cython PACKAGE_VERSION=1.0.0
libusbmuxd-2.0.3/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=2.0.3
libimobiledevice-1.3.1/autogen.sh --prefix=${XXX_PREFIX} --without-cython PACKAGE_VERSION=1.3.1
libirecovery-1.0.1/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=1.0.1
libideviceactivation-1.1.2/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=1.1.2
idevicerestore-1.0.1/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=1.0.1
usbmuxd-1.1.2/autogen.sh --prefix=${XXX_PREFIX} PACKAGE_VERSION=1.1.2