libimobiledevice/libimobiledevice #1465: idevicescreenshot broken with Xcode 15 + iOS 17

lipiz-zhao, 2023-06-28:
Error message when running idevicescreenshot
Failed to capture device screen shot: Could not start screenshotr service: Invalid service
Remember that you have to mount the Developer disk image on your device if you want to use the screenshotr service

But I'm not able to mount the Developer disk image because there's no disk image and signature for iOS 17 in Xcode 15.0's DeviceSupport directory anymore.

Schlaubischlump, 2023-06-28:
iOS 17 changes the way DeveloperDiskImages work. libimobiledevice is not updated yet. Additionally the API for the screenshot relay also [changed](https://github.com/doronz88/pymobiledevice3/issues/471#issuecomment-1596507270). The research is still ongoing, so I assume it will take a while until libimobiledevice is updated.

mds2018, 2023-07-21:
We're having the same issue. Can't take screenshots on iOS 17. Would there be any update as to when the fix could be expected?

remysabliet, 2023-08-16:
In my company we also are waiting for a patch to be release.
宜しくお願いします。

saratchandrakarumuri, 2024-06-20:
Since apple stopped providing Developer disk image from Xcode 15, I don't think there is way to do this going forward. Someone requested for the same in apple's forum [here](https://forums.developer.apple.com/forums/thread/755919). Maybe upvoting that would put it in apple's radar