libimobiledevice/libimobiledevice #1468: idevice_id returns no output and no error

Ben-Voris, 2023-07-05:
# Using `libimobiledevice`

On a system running Ubuntu 18.04, when I plug my iPhone 12 Pro (16.5.1) into a
USB port, the 2017 version of `idevice_id` (it does not accept `-v`) prints
nothing:

```txt
: idevice_id --list --debug ; echo $?
0
```

The new `idevice_id` gives no output, but at least we have a version:

```text
: /usr/local/bin/idevice_id -v
idevice_id 1.3.0-184-g643a46c

: /usr/local/bin/idevice_id --list --debug ; echo $?
0
```

The weird thing to me is that `idevicedate` and `idevicename` both work.

```text
: /usr/local/bin/idevicedate 
Tue Jul  4 18:33:09 MDT 2023
: /usr/local/bin/idevicename 
bvoris’s iPhone 12
```

## Running the new usbmuxd

```shell
export LD_LIBRARY_PATH=/usr/local/lib
sudo -E LD_LIBRARY_PATH=/usr/local/lib /usr/local/lib/usbmuxd -f -v -v
```

### usbmuxd when running running `/usr/local/bin/idevice_id`

```text
[18:27:43.451][4] Client 17 accepted: /usr/lib/upower/upowerd[1128]
[18:27:43.451][5] Client 17 command len 504 ver 1 msg 8 tag 1347
[18:27:43.451][5] Client 17 output buffer got tag 1347 msg 8 payload_length 748
[18:27:43.452][4] Client 17 connection closed
[18:27:43.452][4] Client 17 is going to be disconnected: /usr/lib/upower/upowerd[1128]
[18:27:48.456][4] Client 17 accepted: /usr/lib/upower/upowerd[1128]
[18:27:48.456][5] Client 17 command len 504 ver 1 msg 8 tag 1348
[18:27:48.457][5] Client 17 output buffer got tag 1348 msg 8 payload_length 748
[18:27:48.457][4] Client 17 connection closed
[18:27:48.457][4] Client 17 is going to be disconnected: /usr/lib/upower/upowerd[1128]
```

### usbmuxd debug output from inserting phone to opening phone from file manager

```text
: sudo -E LD_LIBRARY_PATH=/usr/local/lib /usr/local/sbin/usbmuxd -f -v -v
[14:29:41.943][3] usbmuxd v1.1.1-48-g01c94c7 starting up
[14:29:41.943][4] Creating socket
[14:29:41.943][4] Listening on /var/run/usbmuxd
[14:29:41.943][5] Initialized config_dir to /var/lib/lockdown
[14:29:41.943][5] client_init
[14:29:41.943][5] device_init
[14:29:41.943][4] Initializing USB
[14:29:41.943][3] Using libusb 1.0.21
[14:29:41.947][4] Registering for libusb hotplug events
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.003728] [000074a3] libusb: debug [libusb_get_device_list] 
[ 0.003764] [000074a3] libusb: debug [libusb_get_device_descriptor] 
[14:29:41.947][4] Found new device with v/p 05ac:12a8 at 1-12
[... 234 lines cut ...]
[42.894487] [000074a3] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[42.894500] [000074a3] libusb: debug [usbi_handle_transfer_completion] transfer 0x562c76eae340 has callback 0x562c74e57840
[14:30:24.838][5] Device 1-12 RX transfer cancelled
[42.894529] [000074a3] libusb: debug [libusb_free_transfer] transfer 0x562c76eae340
[42.894552] [000074a3] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[42.894573] [000074a3] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[42.894595] [000074a3] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[42.894619] [000074a3] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[42.894640] [000074a3] libusb: debug [usbi_handle_transfer_completion] transfer 0x562c76eb0fd0 has callback 0x562c74e57840
[14:30:24.838][5] Device 1-12 RX transfer cancelled
[42.894683] [000074a3] libusb: debug [libusb_free_transfer] transfer 0x562c76eb0fd0
[42.894708] [000074a3] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[42.894734] [000074a3] libusb: debug [handle_events] poll() returned 0
[42.894756] [000074a3] libusb: debug [libusb_release_interface] interface 1
[42.894806] [000074a3] libusb: debug [libusb_close] 
[42.894831] [000074a3] libusb: debug [usbi_remove_pollfd] remove fd 11
[42.894875] [000074a3] libusb: debug [libusb_exit] 
[42.894889] [000074a3] libusb: debug [libusb_exit] destroying default context
[14:30:24.838][5] device_shutdown
[14:30:24.838][5] client_shutdown
[14:30:24.839][4] Client 12 is going to be disconnected: /usr/lib/gvfs/gvfs-afc-volume-monitor[1386]
[14:30:24.839][3] Shutdown complete
```

### usbmuxd debug output when ejecting phone from file manager

Ben-Voris, 2023-07-05:
Also, `idevicedate` causes the phone to ask if I trust this device. `idevice_id` does not.

mexmer, 2023-07-05:
some information like device model are available without trust, but apple reduced number of things you can get from device without trust to bare minimum, so most information about device requires trust. it gets worse with each ios release.

as for your logs, i suggest removing, or at least stopping upowerd, there has been multiple cases when it did interfere with manually compiled libimobiledevice, because it was compiled against older version.