libimobiledevice/libimobiledevice #1470: Error -38 (MC Protected) - Supervision identity cert ignored?

skulldoggery, 2023-07-08:
Hopefully user error here... 

Running Ubuntu 23.04 on a laptop for mobile testing/recovery purposes. Latest build of `libimobiledevice`

I have:

- an DEP enrolled, MDM managed device
-  a supervision identity cert, added to the MDM

The same device pairs successfully with a Mac host, with the supervision identity cert in Keychain and using Apple Configurator/Console. 

The same cert has been added to the Linux host, but both before doing so and afterward, calling `ideviceinfo` returns a lockdown error of `-38`.  The device ID and name can be interrogated successfully using `idevice_id`

Is there anything obvious I'm missing?

closed 2023-07-08 by skulldoggery

skulldoggery, 2024-02-23:
> Sorry to bump this older issue, I'm having the same issue, did you ever get this to work? How could you add the supervision certificate to lockdownd?

Hmm... Let me go back and look! I'll get back to you! 

The supervision identity is added by your MDM of choice, either Apple Configurator or another product. You would need to create your own, I can share those steps if needed. But the phone would need to be managed/supervised, a personal device should not need any of that

xcbjky, 2024-06-22:
What methods are there to bypass this policy? MDM does not allow the phone to pair with iTunes, returning MC Protect (-38).