libimobiledevice/libimobiledevice #1473: no wifi sync only usb sync

mothabrain, 2023-07-22:
when i connect my iphone directly via usb everything works fine:

idevicebackup2 -d -u 123123123123123123 backup /home/ios/
Backup directory is "/home/ios/"
Started "com.apple.mobilebackup2" service on port 52984.
Negotiated Protocol Version 2.1
Reading Info.plist from backup.
Starting backup...
Backup will be unencrypted.
Requesting backup from device...
Incremental backup mode.

if i try the same with the "-n" switch no device is present

idevicebackup2 -n -d -u 123123123123123123 backup /home/ios/
No device found with udid 123123123123123123

i welcome any hint or solution!

Schlaubischlump, 2023-07-25:
If you use linux you need a version of usbmuxd that supports Wi-Fi devices, e.g. https://github.com/tihmstar/usbmuxd2

olly1240, 2023-08-05:
I found netmuxd to be reliable and is released as a single executable
https://github.com/jkcoxson/netmuxd