libimobiledevice/libimobiledevice #1474: build failure Making all in cython

RFDAJE, 2023-07-22:
I try to build libimobiledevice git on archlinux, but failed with the following error.

```
Making all in cython
make[2]: Entering directory '/home/XXX/.cache/yay/libimobiledevice-git/src/libimobiledevice/cython'
/usr/bin/cython -I/usr/include/plist/cython -I../src -o imobiledevice.c imobiledevice.pyx
/usr/lib/python3.11/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /home/XXX/.cache/yay/libimobiledevice-git/src/libimobiledevice/cython/imobiledevice.pxd
  tree = Parsing.p_module(s, pxd, full_module_name)
warning: imobiledevice.pxd:6:4: Exception already a builtin Cython type

Error compiling Cython file:
------------------------------------------------------------
...
    cdef iDeviceEvent event = iDeviceEvent.__new__(iDeviceEvent)
    event._c_event = c_event
    (<object>user_data)(event)

def event_subscribe(object callback):
    cdef iDeviceError err = iDeviceError(idevice_event_subscribe(idevice_event_cb, <void*>callback))
                                                                 ^
------------------------------------------------------------

imobiledevice.pyx:103:65: Cannot assign type 'void (const_idevice_event_t, void *) except * nogil' to 'idevice_event_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...

    cdef inline BaseError _error(self, int16_t ret):
        return NotificationProxyError(ret)

    cpdef set_notify_callback(self, object callback):
        self.handle_error(np_set_notify_callback(self._c_client, np_notify_cb, <void*>callback))
                                                                 ^
------------------------------------------------------------

notification_proxy.pxi:104:65: Cannot assign type 'void (const_char_ptr, void *) except *' to 'np_notify_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
        else:
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)
        if callback is None:
            err = instproxy_install(self._c_client, pkg_path, c_options, NULL, NULL)
        else:
            err = instproxy_install(self._c_client, pkg_path, c_options, instproxy_notify_cb, <void*>callback)
                                                                         ^
------------------------------------------------------------

installation_proxy.pxi:126:73: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
        else:
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)
        if callback is None:
            err = instproxy_upgrade(self._c_client, pkg_path, c_options, NULL, NULL)
        else:
            err = instproxy_upgrade(self._c_client, pkg_path, c_options, instproxy_notify_cb, <void*>callback)
                                                                         ^
------------------------------------------------------------

installation_proxy.pxi:153:73: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)

        if callback is None:
            err = instproxy_uninstall(self._c_client, appid, c_options, NULL, NULL)
        else:
            err = instproxy_uninstall(self._c_client, appid, c_options, instproxy_notify_cb, <void*>callback)
                                                                        ^
------------------------------------------------------------

installation_proxy.pxi:180:72: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)

        if callback is None:
            err = instproxy_archive(self._c_client, appid, c_options, NULL, NULL)
        else:
            err = instproxy_archive(self._c_client, appid, c_options, instproxy_notify_cb, <void*>callback)
                                                                      ^
------------------------------------------------------------

installation_proxy.pxi:237:70: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)

        if callback is None:
            err = instproxy_restore(self._c_client, appid, c_options, NULL, NULL)
        else:
            err = instproxy_restore(self._c_client, appid, c_options, instproxy_notify_cb, <void*>callback)
                                                                      ^
------------------------------------------------------------

installation_proxy.pxi:265:70: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)

        if callback is None:
            err = instproxy_remove_archive(self._c_client, appid, c_options, NULL, NULL)
        else:
            err = instproxy_remove_archive(self._c_client, appid, c_options, instproxy_notify_cb, <void*>callback)
                                                                             ^
------------------------------------------------------------

installation_proxy.pxi:293:77: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'
make[2]: *** [Makefile:809: imobiledevice.c] Error 1
make[2]: Leaving directory '/home/XXX/.cache/yay/libimobiledevice-git/src/libimobiledevice/cython'
make[1]: *** [Makefile:466: all-recursive] Error 1
make[1]: Leaving directory '/home/XXX/.cache/yay/libimobiledevice-git/src/libimobiledevice'
make: *** [Makefile:396: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

exentio, 2023-08-02:
Can confirm having the same issue

```
Making all in cython
make[2]: Entering directory '/home/ex/.cache/aurutils/sync/libimobiledevice-git/src/libimobiledevice/cython'
/usr/sbin/cython -I/usr/include/plist/cython -I../src -o imobiledevice.c imobiledevice.pyx
/usr/lib/python3.11/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /home/ex/.cache/aurutils/sync/libimobiledevice-git/src/libimobiledevice/cython/imobiledevice.pxd
  tree = Parsing.p_module(s, pxd, full_module_name)
warning: imobiledevice.pxd:6:4: Exception already a builtin Cython type

Error compiling Cython file:
------------------------------------------------------------
...
    cdef iDeviceEvent event = iDeviceEvent.__new__(iDeviceEvent)
    event._c_event = c_event
    (<object>user_data)(event)

def event_subscribe(object callback):
    cdef iDeviceError err = iDeviceError(idevice_event_subscribe(idevice_event_cb, <void*>callback))
                                                                 ^
------------------------------------------------------------

imobiledevice.pyx:103:65: Cannot assign type 'void (const_idevice_event_t, void *) except * nogil' to 'idevice_event_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...

    cdef inline BaseError _error(self, int16_t ret):
        return NotificationProxyError(ret)

    cpdef set_notify_callback(self, object callback):
        self.handle_error(np_set_notify_callback(self._c_client, np_notify_cb, <void*>callback))
                                                                 ^
------------------------------------------------------------

notification_proxy.pxi:104:65: Cannot assign type 'void (const_char_ptr, void *) except *' to 'np_notify_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
        else:
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)
        if callback is None:
            err = instproxy_install(self._c_client, pkg_path, c_options, NULL, NULL)
        else:
            err = instproxy_install(self._c_client, pkg_path, c_options, instproxy_notify_cb, <void*>callback)
                                                                         ^
------------------------------------------------------------

installation_proxy.pxi:126:73: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
        else:
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)
        if callback is None:
            err = instproxy_upgrade(self._c_client, pkg_path, c_options, NULL, NULL)
        else:
            err = instproxy_upgrade(self._c_client, pkg_path, c_options, instproxy_notify_cb, <void*>callback)
                                                                         ^
------------------------------------------------------------

installation_proxy.pxi:153:73: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)
[... 34 lines cut ...]
------------------------------------------------------------

installation_proxy.pxi:265:70: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)

        if callback is None:
            err = instproxy_remove_archive(self._c_client, appid, c_options, NULL, NULL)
        else:
            err = instproxy_remove_archive(self._c_client, appid, c_options, instproxy_notify_cb, <void*>callback)
                                                                             ^
------------------------------------------------------------

installation_proxy.pxi:293:77: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'
make[2]: *** [Makefile:809: imobiledevice.c] Error 1
make[2]: Leaving directory '/home/ex/.cache/aurutils/sync/libimobiledevice-git/src/libimobiledevice/cython'
make[1]: *** [Makefile:466: all-recursive] Error 1
make[1]: Leaving directory '/home/ex/.cache/aurutils/sync/libimobiledevice-git/src/libimobiledevice'
make: *** [Makefile:396: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

harisrahman, 2023-08-03:
i'm having the same issue as well.

crazymind90, 2023-08-05:
In my case , the installed Cython version is 3.0.0 ,
you can check by typing : 
```
cython --version
``` 

the solution is to downgrade it using [ pip ] : 
```
pip install cython==0.29.24
```
it will uninstall 3.0.0 and install 0.29.24 automatically ,

then once again run : 
```
make 
```

deyvs02, 2023-08-09:
I try to compile idevicerestore for windows in exe format and I can't.
I don't think I'm capable, can someone compile it and send it to my email
[deyvs02@hotmail.com](mailto:deyvs02@hotmail.com)

exentio, 2023-08-10:
>can someone compile it and send it to my email

No need to, you can download it from the artifacts on the Actions page, you can trust it more than random people compiling it for you:  
https://github.com/libimobiledevice/libimobiledevice/actions/runs/5721032992#:~:text=libimobiledevice%2Dlatest_x86_64%2Dmingw64

HLFH, 2023-08-18:
`cython0` [is available](https://archlinux.org/packages/extra/x86_64/cython0/) on Arch Linux.
`libimobiledevice` only works with `cython0`, not `cython` or `cython-git`.

closed 2023-08-23 by RFDAJE

thesamesam, 2023-08-28:
@RFDAJE Did you close this just because cython0 is in Arch? If so, that's not a proper fix. Please reopen the bug until the source code is made compatible with Cython 3.

MenacingMight, 2024-01-05:
I am having the same problem. Did anyone solve it?