libimobiledevice/libimobiledevice #1476: 1.3.0 + master: loop in dependencies

kloczek, 2023-08-02:
Looks like it is loop in build dependencies between `libimobiledevice` and `usbmuxd`.
During the autoconf of `usbmuxd` requires `libimobiledevice`
```console
checking pkg-config is at least version 0.9.0... yes
checking for libusb-1.0 >= 1.0.9... yes
checking for libplist-2.0 >= 2.3.0... yes
checking for libimobiledevice-1.0 >= 1.3.0... no
checking for libimobiledevice-glue-1.0 >= 1.0.0... yes
configure: error: preflight worker support requested but libimobiledevice could not be found
```
and `libimobiledevice` requires `usbmuxd`
```console
checking whether the /usr/bin/g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for x86_64-redhat-linux-gnu-pkg-config... no
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusbmuxd-2.0 >= 2.0.2... no
configure: error: Package requirements (libusbmuxd-2.0 >= 2.0.2) were not met:

Package 'libusbmuxd-2.0', required by 'virtual:world', not found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libusbmuxd_CFLAGS
and libusbmuxd_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```
I don't see in autoconf options how to break temporary that build dependency loop.
Any hints? 🤔

mexmer, 2023-08-02:
@kloczek libimobiledevice requires libubsmuxd, eg. usbmux client library, not usbmuxd, whichs is usbmux daemon
it's two separate projects
libusbmuxd - https://github.com/libimobiledevice/libusbmuxd
usbmuxd - https://github.com/libimobiledevice/usbmuxd

kloczek, 2023-09-30:
BTW .. is it any plan to make the new release soon? 🤔 
Last one was +3.5 years ago and looking on https://github.com/libimobiledevice/libimobiledevice/compare/1.3.0...master there are already ~190 commits added since last release