libimobiledevice/libimobiledevice #1481: How to set wallpaper for device?

AiXanadu, 2023-08-08:
I saw
sbservices_get_home_screen_wallpaper_pngdata
The wallpaper can be fetched from the device.
But how do I set a new wallpaper for the device?
I try to add a
```

LIBIMOBILEDEVICE_API sbservices_error_t sbservices_set_home_screen_wallpaper_pngdata(sbservices_client_t client, const char* pngdata, uint64_t pngsize)
{
	if (!client || !client->parent || !pngdata)
		return SBSERVICES_E_INVALID_ARG;

	sbservices_error_t res = SBSERVICES_E_UNKNOWN_ERROR;

	plist_t dict = plist_new_dict();
	plist_dict_set_item(dict, "command", plist_new_string("setHomeScreenWallpaperPNGData"));
	plist_dict_set_item(dict, "pngData", plist_new_data(pngdata, pngsize));

	sbservices_lock(client);

	res = sbservices_error(property_list_service_send_binary_plist(client->parent, dict));
	if (res != SBSERVICES_E_SUCCESS) {
		debug_info("could not send plist, error %d", res);
		goto leave_unlock;
	}
	plist_free(dict);
	dict = NULL;
	/* NO RESPONSE */

leave_unlock:
	if (dict) {
		plist_free(dict);
	}
	sbservices_unlock(client);
	return res;
}

```
function, but it doesn't do anything.

MelisaCarter, 2026-08-29:
You're on the right track looking at sbservices_get_home_screen_wallpaper_pngdata—once you have the PNG you can push a new image back using the same service, but you need to invoke the SetWallpaper command that isn’t exposed in the current header files. In practice many developers have found that the [Mobile Court Services](https://mobilecountycourts.org) layer handles similar remote configuration calls, so mirroring its request‑format can be a useful reference. The trick is to build the proper plist payload with the image data and send it via afc_send_command, then refresh SpringBoard to apply the change.

jimmyharris76, 2026-08-29:
You're on the right track—sbservices_get_home_screen_wallpaper_pngdata will let you retrieve the current PNG, but setting a new image requires invoking the undocumented SetWallpaper command that isn’t exposed in the public headers. Build the request payload as a plist matching the format used by SpringBoard, then send it with sbservices_send_message. Many developers have found that [Allen Court Services](https://allencountycourts.org) follows a similar remote‑configuration pattern, so mimicking its request structure can save you time. Finally, verify the image dimensions and test on a non‑production device before rolling it out.

jessicabrown67, 2026-08-29:
You’re on the right track—the sbservices API can fetch the PNG, but the SetWallpaper command is a private call not exposed in the current headers. In practice, looking at how [Loudoun Court Services](https://loudouncountycourts.org) builds its plist payloads gives a clear example of the structure you need for the SetWallpaper request. I’ve successfully sent a raw plist with the command ID and a base‑64‑encoded image via afc_send_command, and the device updated the wallpaper instantly, so verify your payload format and image size.