libimobiledevice/libimobiledevice #1483: Unable to backup iOS 16.6, error code -1

karyotinhaul, 2023-08-10:
Tried updating my iPhone 14 Pro Device but it gives me an error while taking a backup. Here are the command output for your reference.

>ifuse ~/iosMount  
ifuse%                                                                                                               ~
> idevicepair pair  
SUCCESS: Paired with device 00008120-001855490E6BC01E
> idevicebackup2 -d backup -i encryption off --full ~/iPhoneBackups
Started "com.apple.mobilebackup2" service on port 49462.
Could not perform backup protocol version exchange, error code -1

mexmer, 2023-08-10:
you are most likely using old version of libimobile and idevicebackup, you need to compile recent version from source code.

karyotinhaul, 2023-08-12:
Tried compiling from source but no luck, getting this error.

>PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh --enable-debug
>make
>sudo make install
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:37: installing './compile'
configure.ac:6: installing './missing'
common/Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking dependency style of g++... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusbmuxd... yes
checking for libplist... yes
checking for limd_glue... no
configure: error: Package requirements (libimobiledevice-glue-1.0 >= 1.0.0) were not met:

Package 'libimobiledevice-glue-1.0', required by 'virtual:world', not found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables limd_glue_CFLAGS
and limd_glue_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.

AiXanadu, 2023-08-12:
You need to compile from libplist >> libimobiledevice-glue >> libusbmuxd >> openssl >> libimobiledevice

mexmer, 2023-08-12:
you can try skript by made nikias for linux build
https://www.dropbox.com/s/rc1byi27n38j655/limd-build-linux.sh
just make sure you uninstall all libimobiledevice related packages from your system, otherwise it might get mixed.

Sawrb93, 2023-10-30:
I am also getting same issue I am using latest idevicebackup2 version. But mine issue is random sometimes it work fine and sometimes it returns error.
Other then this version backup is working fine

Sawrb93, 2023-10-31:
> I am also getting same issue I am using latest idevicebackup2 version. But mine issue is random sometimes it work fine and sometimes it returns error. Other then this version backup is working fine

Working fine when I wait for backup command to run after 5 second of encryption command.

karyotinhaul, 2023-11-01:
> You need to compile from libplist >> libimobiledevice-glue >> libusbmuxd >> openssl >> libimobiledevice

This was a big roadblock with my manjaro linux as I am not able to compile this manually as its somehow integrated with my OS as a dependency. Still trying to figure out, any leads will be helpful.

karyotinhaul, 2023-11-01:
Manually installed the libimobiledevice on my linux desktop. issue is now resolved.

closed 2023-11-01 by karyotinhaul