libimobiledevice/libimobiledevice #1487: does this lib work on termux?

ben1681, 2023-08-25:
I found this pkg in termux, then after I installed it and its dependencies in my android phone,linked my iphone with OTG line and type cmd like idevieid,i got the response"ERROR: No device found!". I installed termux:API and tried again, it still didn't work.  does this lib work on termux? or what else should I do? Is root needed? Very grateful to the people who can help.

mexmer, 2023-08-25:
libimobiledevice is just client library for communication, you need usbmuxd running to be able to detect and communicate with connected ios device. if there is termux package with ubsmuxd, i have no idea.

ben1681, 2023-08-25:
> libimobiledevice is just client library for communication, you need usbmuxd running to be able to detect and communicate with connected ios device. if there is termux package with ubsmuxd, i have no idea.

thank you for help, termux also has usbmuxd in it, when install libimobiledevice, usbmuxd installed too. Could you tell me how to start usbmuxd?

mexmer, 2023-08-25:
or normal linux usbmuxd is either started by systemd or udevd when device is plugged in, or is run in daemon mode from init scripts.

when i debug stuff, i usually run it in foreground 
```
usbmuxd -f
```
but in termux, it really depends, where you have write access and such, you might need to set path to pidfile, logfile, also must have access to usb devices under user that is running ubsmuxd.

you can see all parameters by running
```
usbmuxd -h
```
never tried this on android, so no idea, how to run it in user mode.

ben1681, 2023-08-28:
@mexmer thanks a lot.It's not easy to make it work on android, I need to learn a lot first.