libimobiledevice/libimobiledevice #1490: Current status of the support for iOS 17

mforys, 2023-09-07:
Hi!
I'm a bit worried about the upcoming iOS 17 release. 
What is the status of work on the support for it in **libimobiledevice** community?
I'm asking because if there is no such support even in plans then the switch to another solution is needed on my side.

Cheers!

mexmer, 2023-09-07:
libimobiledevice works with ios 17, only issue i know about is developer image mount and connection to developer console. as for other functions so far, i had not found any issue (sync works, afc works, deviceinfo works, pairing/unpairing works, backup works).
to me, except access to developer functionality, apple didn't do any significant changes to how ios/ipados communicates.

Schlaubischlump, 2023-09-10:
Location simulation is broken as well, even if the mounting of personalized developer disk images is fixed.

lupenkonik, 2023-09-13:
Hi everybody, was there by chance any workaround/fix for idevicescreenshot iOS 17.x support?

Thanks in advance

Schlaubischlump, 2023-09-13:
I think pymobiledevice3 supports taking screenshots on iOS 17.

lupenkonik, 2023-09-13:
> I think pymobiledevice3 supports taking screenshots on iOS 17. 

Thanks, I'll take a look on that.

mexmer, 2023-09-14:
you can check for more info here
https://github.com/doronz88/pymobiledevice3

author did python implementation of RemoteXPC protocol (used by xcode for communication with developer tools in phone) and for mounting "personalized" disk images.

luc-git, 2023-09-30:
> libimobiledevice works with ios 17, only issue i know about is developer image mount and connection to developer console. as for other functions so far, i had not found any issue (sync works, afc [...]

Hello have you tested those functionnalities in the latest ios version? Do house_arrest and SpringBoardServices (com.apple.springboardservices) work as well?

mexmer, 2023-10-02:
house arrest works, as for springboard, we are only using it to grab wallpaper, and that works too, no idea about other functions of springboard.

HaoWShi, 2023-10-17:
I found an issue that I can not get device info by idevice_id -l or -n when I make my iphone and mac connect to the wifi and config config via network in xcode. 

Mac::2019 MacBook Pro, MacOS Sonoma 14.0
Xcode: 15.0
Iphone: Iphone14 Pro ,IOS17.0.3

ravindra1307, 2023-10-26:
Facing issue in detecting devices over network. 
Apple TV : tvOS 17.0
Xcode : 15.0

ankasani, 2023-10-27:
Hi , Is there any workaround/fix for  iOS 17.x support?

wesley6688, 2024-01-20:
Facing the same issue
**idevice_id -n** does not detect the iPhone 15 with iOS 17.3 connected with WIFI.
macOS: 14.2
but,
iPhone 13 with iOS 17.2 works fine over same WIFI setup.

tja, 2024-03-06:
Any news on this? Facing the same issue.

Schlaubischlump, 2024-03-06:
I don't think there is anything new for libimobiledevice, but there is pymobiledevice3 that works for iOS 17 and go-ios which almost finished implementing iOS 17 support.

wesley6688, 2024-03-07:
Take a look at this post, it should resolve the issue.
https://github.com/libimobiledevice/libimobiledevice/issues/1497

tja, 2024-03-07:
@wesley6688 Thank you so much. Yeah, that did it!