libimobiledevice/libimobiledevice #1491: Could not connect to lockdownd: Mux error (-8)

haasn, 2023-09-14:
I'm on the iOS 17 RC (Public Beta), and trying to re-establish an iFuse connection. It was working fine earlier today, until the process got interrupted while transferring files (as usually happens several times during a full backup).

Several reboots of my PC and phone later, I can no longer achieve even a basic connection. Here is the observed behavior:

### Reproduction steps

1. `systemctl mask usbmuxd.service` (for debugging purposes)
2. `sudo usbmuxd -fuvv`
3. plug in iPhone and unlock
4. `idevice_id -l` -> No output, `ideviceinfo` -> no device found
5. Wait maybe 30-60 seconds
6. `idevice_id -l` -> Starts showing output
7. `ideviceinfo --simple` -> ERROR: Could not connect to lockdownd: Mux error (-8)

If I omit `--simple` then I get a trust dialog, sometimes. I have yet to figure out why it takes upwards of 30s to establish a connection, and sometimes upwards of 30s just to show the trust dialog. Accepting this dialog does nothing.

### Additional information

I wish I could provide a more coherent bug report, but I have tried every possible variation of these steps and it seems that every time I run into a different bug. But no matter what I try, it seems I cannot ever establish another connection on this device. It seems cursed / doomed for failure.

Which is just frustrating given that I'm midway through an rsync backup that was running just fine up until I did something (opening the "account recovery" settings in the settings app) that interrupted it.

### Full log

I followed the above steps, except repeating steps `6` and `7` a few times (and then at the end, without `--simple`, although for some reason this time I *didn't* see a trust dialog...)

<details>
<summary>Click to expand...</summary>
<p>
```
[22:21:43.615][3] usbmuxd v1.1.1+git48.01c94c7 starting up
[22:21:43.615][4] Creating socket
[22:21:43.615][4] Listening on /var/run/usbmuxd
[22:21:43.615][5] Initialized config_dir to /var/lib/lockdown
[22:21:43.615][5] client_init
[22:21:43.615][5] device_init
[22:21:43.615][4] Initializing USB
[22:21:43.615][3] Using libusb 1.0.26
[22:21:43.631][4] Registering for libusb hotplug events
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.015403] [0000660f] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x563c2a976dc0 with handle 1
[ 0.015410] [0000660f] libusb: debug [libusb_get_device_list]  
[ 0.015420] [0000660f] libusb: debug [libusb_get_device_descriptor]  
[ 0.015425] [0000660f] libusb: debug [libusb_get_device_descriptor]  
[22:21:43.631][4] Found new device with v/p 05ac:12a8 at 1-12
[ 0.015438] [0000660f] libusb: debug [libusb_open] open 1.12
[ 0.015470] [0000660f] libusb: debug [usbi_add_event_source] add fd 9 events 4
[22:21:43.631][4] Requesting current mode from device 1-12
[ 0.015482] [0000660f] libusb: debug [libusb_submit_transfer] transfer 0x563c2a978d30
[ 0.015488] [0000660f] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[22:21:43.631][4] 1 device detected
[22:21:43.631][3] Initialization complete
[22:21:43.631][3] Automatic device discovery on hotplug disabled.
[22:21:43.631][5] usb polling enable: 0
[22:21:43.631][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
[ 0.015536] [0000660f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.015543] [0000660f] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.015549] [0000660f] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 0ms
[ 0.015555] [0000660f] libusb: debug [usbi_wait_for_events] poll() returned 0
[ 0.019441] [0000660f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.019454] [0000660f] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 0ms
[ 0.019460] [0000660f] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.019468] [0000660f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.019472] [0000660f] libusb: debug [handle_control_completion] handling completion status 0
[ 0.019476] [0000660f] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.019482] [0000660f] libusb: debug [usbi_handle_transfer_completion] transfer 0x563c2a978d30 has callback 0x563c2a3be5b0
[22:21:43.635][4] Received response 3:3:3:0 for get_mode request for device 1-12
[22:21:43.635][2] Skipping switch device 1-12 mode from 1 to 1
[ 0.019497] [0000660f] libusb: debug [libusb_get_configuration]  
[... 545 lines cut ...]
[436.048045] [0000660f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[436.048051] [0000660f] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 0ms
[436.048057] [0000660f] libusb: debug [usbi_wait_for_events] poll() returned 1
[436.048064] [0000660f] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=36
[436.048069] [0000660f] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[436.048073] [0000660f] libusb: debug [handle_bulk_completion] all URBs in transfer reaped --> complete!
[436.048077] [0000660f] libusb: debug [usbi_handle_transfer_completion] transfer 0x563c2a9a89e0 has callback 0x563c2a3b73c0
[436.048081] [0000660f] libusb: debug [libusb_free_transfer] transfer 0x563c2a9a89e0
[22:29:04.668][4] Client 10 connection closed
[22:29:04.668][4] Client 10 is going to be disconnected: /proc/28289/cmdline[28289]
[22:29:04.668][4] Client died mid-connect, aborting device 1 connection
[22:29:04.668][5] connection_teardown dev 1 sport 4 dport 62078
[22:29:04.668][5] [OUT] dev=1 sport=4 dport=62078 seq=0 ack=0 flags=0x4 window=131072[512] len=0
[441.052973] [0000660f] libusb: debug [libusb_submit_transfer] transfer 0x563c2a9b9430
[441.052979] [0000660f] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 36
[441.053049] [0000660f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[441.053054] [0000660f] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 0ms
[441.053058] [0000660f] libusb: debug [usbi_wait_for_events] poll() returned 1
[441.053064] [0000660f] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=36
[441.053067] [0000660f] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[441.053071] [0000660f] libusb: debug [handle_bulk_completion] all URBs in transfer reaped --> complete!
[441.053074] [0000660f] libusb: debug [usbi_handle_transfer_completion] transfer 0x563c2a9b9430 has callback 0x563c2a3b73c0
[441.053078] [0000660f] libusb: debug [libusb_free_transfer] transfer 0x563c2a9b9430
```
</p></details>

haasn, 2023-09-14:
Also, for some reason, `ideviceinfo --debug` does nothing - it does not produce any additional output.

haasn, 2023-09-14:
Maybe relevant, but the above all happens if I plug in over USB 2.0. If I plug in over a USB 3.0 port, it somehow gets stuck after the first few messages, and then never does anything else:

```
[Sep14 22:47] usb 1-4: new high-speed USB device number 5 using xhci_hcd
[  +0.198420] usb 1-4: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice=15.02
[  +0.000007] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000004] usb 1-4: Product: iPhone
[  +0.000003] usb 1-4: Manufacturer: Apple Inc.
[  +0.000003] usb 1-4: SerialNumber: 0000812000092D910EDB401E
```

```
λ usbmuxd -fuvvv
[22:46:45.037][3] usbmuxd v1.1.1+git48.01c94c7 starting up
[22:46:45.037][4] Creating socket
[22:46:45.037][4] Listening on /var/run/usbmuxd
[22:46:45.037][5] Initialized config_dir to /var/lib/lockdown
[22:46:45.037][5] client_init
[22:46:45.037][5] device_init
[22:46:45.037][4] Initializing USB
[22:46:45.037][3] Using libusb 1.0.26
[22:46:45.050][4] Registering for libusb hotplug events
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.013053] [00001641] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x55f14a247dc0 with handle 1
[ 0.013082] [00001641] libusb: debug [libusb_get_device_list]  
[ 0.013103] [00001641] libusb: debug [libusb_get_device_descriptor]  
[22:46:45.050][4] 0 devices detected
[22:46:45.050][3] Initialization complete
[22:46:45.050][3] Automatic device discovery on hotplug disabled.
[22:46:45.050][5] usb polling enable: 0
[22:46:45.050][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
[ 0.013197] [00001641] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.013213] [00001641] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.013221] [00001641] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 0ms
[ 0.013227] [00001641] libusb: debug [usbi_wait_for_events] poll() returned 0

(here I plugged in the device)

[17.958085] [00001642] libusb: debug [linux_get_device_address] getting address for device: 1-4 detached: 0
[17.958113] [00001642] libusb: debug [linux_get_device_address] scan 1-4
[17.958159] [00001642] libusb: debug [linux_get_device_address] bus=1 dev=5
[17.958163] [00001642] libusb: debug [udev_hotplug_event] udev hotplug event. action: add.
[17.958165] [00001642] libusb: debug [linux_enumerate_device] busnum 1 devaddr 5 session_id 261
[17.958168] [00001642] libusb: debug [linux_enumerate_device] allocating new device for 1/5 (session 261)
[17.958199] [00001642] libusb: debug [linux_get_parent_info] dev 0x7fae88002d00 (1-4) has parent 0x55f14a2486b0 (usb1) port 4
[17.958229] [00001641] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[17.958239] [00001641] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 0ms
[17.958249] [00001641] libusb: debug [usbi_wait_for_events] poll() returned 1
[17.958255] [00001641] libusb: debug [handle_event_trigger] event triggered
[17.958260] [00001641] libusb: debug [handle_event_trigger] hotplug message received
[17.976394] [00001642] libusb: debug [linux_get_device_address] getting address for device: 1-4 detached: 0
[17.976409] [00001642] libusb: debug [linux_get_device_address] scan 1-4
[17.976487] [00001642] libusb: debug [linux_get_device_address] bus=1 dev=5
[17.976495] [00001642] libusb: debug [udev_hotplug_event] udev hotplug event. action: bind.
```

haasn, 2023-09-14:
I tried plugging my phone directly to a USB-C port instead of using an adapter to USB 3.0 or USB 2.0, and this is what happened: (This time via `usbmuxd.service`)

```
Sep 14 22:55:49 xor systemd[1]: Started Socket daemon for the usbmux protocol used by Apple devices.
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.039][3] usbmuxd v1.1.1+git48.01c94c7 starting up
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.040][3] Successfully dropped privileges to 'usbmux'
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.040][3] Using libusb 1.0.26
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.048][3] Initialization complete
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.048][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.052][2] Skipping switch device 1-8 mode from 1 to 1
Sep 14 22:55:50 xor usbmuxd[11763]: libusb: warning [op_get_configuration] device unconfigured
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.052][3] Found usbmux interface for device 1-8: 1
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.052][3] Changing configuration of device 1-8: 0 -> 4
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.097][3] Connecting to new device on location 0x10008 as ID 1
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.097][3] Connected to v2.0 device 1 on location 0x10008 with serial number 00008120-00092D910EDB401E
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.100][1] RX transfer stalled for device 1-8
Sep 14 22:55:50 xor usbmuxd[11763]: [22:55:50.100][3] Removed device 1 on location 0x10008
```

I've never even seen that error before (RX transfer stalled). This issue is cursed.

haasn, 2023-09-14:
It gets from bad to worse. After running the "Reset Location & Privacy Settings" option on the iPhone, in a vain attempt to get it to "forget" my PC and let me re-trust/pair it properly, I can no longer get a connection whatsoever.

Now it's always stuck on `no device found` again. What a headache. I was also planning on trying git master but I guess I can no longer do that.

Maybe it's time to just throw this phone out of the window, it seems to create nothing but stress.

haasn, 2023-09-14:
Update with some good news: I plugged the phone into my laptop instead of my desktop PC and everything works perfectly as expected. I immediately was able to trust the PC, `ideviceinfo` worked out of the box, and I could pair and mount my data with `ifuse`.

I guess I will now just always have to rsync my photos via ssh to my laptop. So it seems that, whatever the issue is, for some reason my desktop PC is just cursed.

pwaller, 2024-02-26:
I'm experiencing the same symptoms and had a very similar overall experience to you. I too have a (2017) laptop where it works and a (2023) desktop where it doesn't.

The version I'm using for both functioning/non-functioning tests is the one shipped in nixpkgs, `libimobiledevice-1.3.0+date=2023-04-30` (860ffb707af3af94467d2ece4ad258dda957c6cd). I have verified that this version works on one set of hardware but not the other.

I note that the newer hardware is quite a bit faster than the older hardware, so one hypothesis as to the difference/cause could be a timing issue? Another hypothesis is some other kind of hardware related issue.

Another problem I've seen is `LIBUSB_ERROR_NO_DEVICE` emitted by usbmuxd immediately when the device is plugged in. Perhaps suggestive that usbmux is proceeding to do something before the device is visible.

bmenrigh, 2024-12-05:
I'm a bit late to the party here but I was also getting `ERROR: Could not connect to lockdownd: Mux error (-8)` and usbmuxd would show `Client died mid-connect, aborting device 1 connection`.

I was using a USB C cable to a USB C port on my iPhone 16.

Curiously `idevice_id` would work.

Simply moving to USB C to A cable on a different USB port has eliminated the usbmuxd error and `ideviceinfo` now prints out a bunch of details.

I'm not sure if I should blame USB C, or my cable, or some combination of the two but fortunately this other cable is working fine.

borolo222, 2024-12-21:
Had the same issue. Tested with my son's and wife's iphone and worked as expected. I believe it has be an iphone setting thing. I migrated from the problematic iphone to another one and the problem persisted. I then Reset Location & Privacy Settings. And it is working now. I did erase the SystemConfiguration.plist file in /var/lib/lockdown first.

Hope this helps