libimobiledevice/libimobiledevice #1495: Not able to restore contacts

Sawrb93, 2023-10-06:
I am using idevicebackup2 and it is not restoring contact, bookmark and apps installed. Only restoring photos n images.

mexmer, 2023-10-08:
on what device, and what ios version.

for contacts, idevicebackup (or you can say backup agent in device), only restores contacts stored inside device memory. contacts that were synced from google, icloud, and other services will be synced automatically, when you reconnect service (by default if you restoring to same device, it should also restore sync account settings, although you might need to enter password again ... depending on how old backup is)
for apps, you should see placeholder icons, apps will work after your devices has correctly setup apple account, that was used to download and install apps, until then, you will only see placeholder icons.

Sawrb93, 2023-10-09:
Restoring data from iPhone 11(version 16.6) to iPhone SE (3rd Gen) (version 16.6.1) photos and video are restoring but contacts are not. Checked with memory contacts too.

But when I am restoring same backup with iTunes it is restoring contacts(Means our backup have contact files but during restore contacts are not restoring).

Command for backup - 
idevicebackup2 -u sourceDeviceID backup --full backupPath

Command for restoring - 
idevicebackup2 restore --system --settings -s sourceDeviceID -u detinatioDeviceID backupPath

mexmer, 2023-10-10:
do you have backup password enabled? from my experience unecrypted backup contains less info, than encrypted, although i think contacts should be available even in unecrypted one.

other than that. no idea, whald could be wrong.