libimobiledevice/libimobiledevice #1496: IOS 17

SANSA27, 2023-10-10:
Is there any possibility that I can get the developer disk image for ios-17 or do I need to mount the Personalized Developer Disk Image using pymobiledevice , if I have to mount how can I do that?

Schlaubischlump, 2023-10-12:
iOS 17 only allows mounting personalized developer disk images. There are no "normal" developer disk images anymore. Libimobiledevice is not updated to support iOS 17 developer features, since it does not yet implement the new remote xpc API. 

To my knowledge pymobiledevice3 is the only library that supports both of these features.

KingRat0011, 2023-10-17:
Hi, I'm having the same issues
Platform - M2 Mac, Sonoma 14.0

Python 3.12.0

```
diginetiot@MacMini Desktop % sudo python3 -m pymobiledevice3 developer dvt screenshot test.png
[?] choose device: <PlistUsbmuxLockdownClient ID:00008110-0006394E22C3401E VERSION:17.0.3 TYPE:iPhone14,7 PAIRED:True>
<PlistUsbmuxLockdownClient ID:51e86d44707a71cb1774eca3a73f05bf34d0265d VERSION:15.7.9 TYPE:iPhone9,3 PAIRED:True>
❯ <PlistUsbmuxLockdownClient ID:00008110-0006394E22C3401E VERSION:17.0.3 TYPE:iPhone14,7 PAIRED:True>

 

2023-10-17 08:50:21 MacMini __main__[50037] ERROR Failed to access an invalid lockdown service, possibly from DeveloperDiskImage.dmg or a Cryptex. You may try: python3 -m pymobiledevice3 mounter auto-mount
diginetiot@MacMini Desktop % python3 -m pymobiledevice3 mounter auto-mount
[?] choose device: <PlistUsbmuxLockdownClient ID:00008110-0006394E22C3401E VERSION:17.0.3 TYPE:iPhone14,7 PAIRED:True>
<PlistUsbmuxLockdownClient ID:51e86d44707a71cb1774eca3a73f05bf34d0265d VERSION:15.7.9 TYPE:iPhone9,3 PAIRED:True>
❯ <PlistUsbmuxLockdownClient ID:00008110-0006394E22C3401E VERSION:17.0.3 TYPE:iPhone14,7 PAIRED:True>

 

2023-10-17 08:50:42 MacMini pymobiledevice3.cli.mounter[50040] ERROR DeveloperDiskImage already mounted
diginetiot@MacMini Desktop % python3 --version
Python 3.12.0
diginetiot@MacMini Desktop %

 

diginetiot@MacMini Desktop % python3 -m pymobiledevice3 mounter list
[?] choose device: <PlistUsbmuxLockdownClient ID:00008110-0006394E22C3401E VERSION:17.0.3 TYPE:iPhone14,7 PAIRED:True>
<PlistUsbmuxLockdownClient ID:51e86d44707a71cb1774eca3a73f05bf34d0265d VERSION:15.7.9 TYPE:iPhone9,3 PAIRED:True>
❯ <PlistUsbmuxLockdownClient ID:00008110-0006394E22C3401E VERSION:17.0.3 TYPE:iPhone14,7 PAIRED:True>

 

[
    {
        "BackingImage": "/private/var/containers/Shared/SystemGroup/DD8C1AB9-B552-4B4F-9344-AF4A26A0C894/backingStore/69A92701E8E85A3B62FF8EE74F15BD69F3888A61100CB5DD/CA8121833278FD0E3109750496EFB9F66C879EEA0892D4F2/07eWmf.dmg",
        "DeviceNode": "/dev/disk3",
        "DeviceType": "DiskImage",
        "DiskImageType": "Personalized",
        "FilesystemType": "hfs",
        "ImageSignature": "69a92701e8e85a3b62ff8ee74f15bd69f3888a61100cb5ddca8121833278fd0e3109750496efb9f66c879eea0892d4f2",
        "IsMounted": true,
        "IsReadOnly": true,
        "MountPath": "/System/Developer",
        "PersonalizedImageType": "DeveloperDiskImage",
        "PersonalizedImageVersion": "348.1",
        "PersonalizedImageVersionInfo": {
            "BuildVersion": "1266",
            "Platform": "iOS",
            "ProductBuildVersion": "15A240d",
            "ProjectMetadata": [
                {
                    "Project": "XCTest",
                    "Version": "22189"
                },
                {
                    "Project": "Mercury",
                    "Version": "46"
                },
                {
                    "Project": "JetsamProperties",
                    "Version": "1802"
                },
                {
                    "Project": "GPUToolsDevice_DDI",
                    "Version": "230.35"
                },
                {
                    "Project": "DTOCMock",
                    "Version": "22001"
                },
                {
                    "Project": "DTDeveloperDiskImageSupport",
                    "Version": "14.0.0"
                },
                {
                    "Project": "CoreDevice",
                    "Version": "348.1"
                }
            ],
            "Variant": "Public"
        },
        "SupportsContentProtection": false
    }
]

 

diginetiot@MacMini Desktop %
```