libimobiledevice/libimobiledevice #1497: idevice_id can not found Iphone(IOS 17)  via wifi

HaoWShi, 2023-10-17:
I set up a local development environment.
Mac::2019 MacBook Pro, MacOS Sonoma 14.0
Xcode: 15.0
Iphone: Iphone14 Pro ,IOS17.0.3
Appium2

I can run test case in appium when iphone connect to mac by usb. And I make iphone and mac connect to same Wifi(other iphone's wifi hotspot). Then I disconnect usb between Mac and Iphone, XCode connect to Iphone via Wifi successfully

But I run command 'idevice_id -n' and 'idevice_id -l' can not find the iphone connect to mac via wifi

ravindra1307, 2023-10-26:
I am also facing same issue. for me Apple Tv is not getting detected (WIFI network). earlier it was working fine.

ltDino, 2023-11-22:
I am also facing same issue too, you can run 'xctrace list devices' to find the iphone connect to mac via wifi

wesley6688, 2024-01-25:
hi, @HaoWShi 

I running into same issue, but it solved for me when I select the option "Show this iPhone when on WI-FI" in the Finder window when you connect your iPhone with USB cable.
when you connect you iPhone again with WIFI, it will show up when you run the command "idevice_id -n"
Appium will find the device.
Give a try!
[image]

mexmer, 2024-01-25:
yes, it's because that's security feature of ios, nothing to do with libimobiledevice. if wifi sync is not allowed, libimobiledevice will not find ios device over wifi. on OSx libimobiledevice asks operating system service, to get list of connected iOS devices.

kevinlinxc, 2024-03-05:
> hi, @HaoWShi
> 
> I running into same issue, but it solved for me when I select the option "Show this iPhone when on WI-FI" in the Finder window when you connect your iPhone with USB cable. when you connect you iPhone again with WIFI, it will show up when you run the command "idevice_id -n" Appium will find the device. Give a try! ![Screenshot 2024-01-25 at 9 53 48 AM](https://private-user-images.githubusercontent.com/157198820/299761617-a7975cec-6258-411e-a41f-3f3cc6c3ee0a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk2MTg1MzQsIm5iZiI6MTcwOTYxODIzNCwicGF0aCI6Ii8xNTcxOTg4MjAvMjk5NzYxNjE3LWE3OTc1Y2VjLTYyNTgtNDExZS1hNDFmLTNmM2NjNmMzZWUwYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNVQwNTU3MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMjE3MDE5ODJhNDVkZWZkODViNjgyODNiZjYzMWZiMjEwNjc0NWFiYjM0NmE1NzAwMzkxMzA2NmJmMjA5OGIzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.q3KauPnAYcHbZgsyz8UF3ulCpn90Efmwp5xHHW5cVuQ)

This just saved me, thank you!

victorchentw, 2024-07-10:
@wesley6688 
Thank you, it works for iOS 18 too. You saved my day!!

mexmer, 2026-01-04:
this works only on macos, and windows using apple service, on linux you need to have install [ubsmuxd2](https://github.com/jkcoxson/netmuxd) , which is variant of usbmuxd, that has network connection support.