libimobiledevice/libimobiledevice #1502: Get the Unix timestamp in milliseconds

leodeve, 2023-10-31:
Hello,

Firstly, I would like to thank you for all the work done on this libimobiledevice project.
It is very useful to be able to control an iPhone from a Linux PC.
I would like to synchronize the iPhone and the PC thanks to the Unix timestamp in milliseconds.
I tried the `idevicedate` command. However, it shows me the date with seconds precision.
I checked the --sync option but it uses the time function which returns the Unix timestamp in seconds.
Is it possible to add a command called `idevicetimestamp` which returns the Unix timestamp in milliseconds ?
The following code seems to work :

```c
#include <stdio.h>
#include <sys/time.h>
#include <stdint.h>
  
int main()
{
    struct timeval tv;
    gettimeofday(&tv, NULL);
    int64_t time_in_millis = (tv.tv_sec) * 1000 + (tv.tv_usec) / 1000;
    printf("%ld", time_in_millis); // print 1698749623875

    return 0;
}
```

lstarboy, 2023-11-01:
ideviceinfo -k TimeIntervalSince1970 return the Unix timestamp in seconds with float type

leodeve, 2023-11-02:
Thanks a lot! That’s what I was looking for.

closed 2023-11-02 by leodeve