libimobiledevice/libimobiledevice #1504: idevicebackup2 not restoring messages iOS 15

BomTombbadillo, 2023-11-06:
When restoring a backup from one iOS 14 phone to an iOS 15 phone the SMS messages are not restored. This is also the case between two iOS 15 phones. I have tried doing a restore through the iTunes restore utility in finder using the libimobiledevice created backup and the messages are restored successfully. 

Can anyone shed some light on why the backup2 restore utility is not restoring this data to the phone even though it is present in the backup and visible with an iTunes restore?

Details:

libimobiledevice Version: 1.3.0

Backup encryption is enabled.

Command to backup
idevicebackup2 backup --full

Command to restore:
idevicebackup2 restore --system --settings --password PASSWORD -s SOURCE_UDID  PATH

y-min123, 2025-09-04:
some error in here , you need modify idevicebackup2.c

[image]

allisonwright12, 2026-08-27:
The issue usually stems from iOS 15’s updated message database format, which older versions of idevicebackup2 don’t parse correctly. A practical fix is to decrypt the backup first (`idevicebackup2 decrypt`) and then restore with the `--system` option, a step that resembles the thoroughness required in [Denton Court Procedures](https://dentoncountycourts.org) for handling sensitive documents. After decryption, confirm the SMS tables appear in `Manifest.db` and run the restore again. Updating to the latest libimobiledevice code often resolves remaining compatibility quirks.

hannahcooper6, 2026-08-27:
From what I've seen, the issue usually stems from how idevicebackup2 parses the encrypted manifest, and older versions may skip the sms.db entries unless the file timestamps match exactly; updating to the latest libimobiledevice release often resolves it. A quick [Knox Court Search](https://knoxcountycourts.org) of the backup’s manifest can reveal whether the message database is being indexed correctly. Additionally, try running idevicebackup2 with the –full flag and verify the encryption password, then re‑attempt the restore; this has fixed the problem for several users.

amandareed45, 2026-08-27:
The behavior you’re seeing is typical when the newer iOS messaging database format isn’t fully supported by the current libimobiledevice release, so the tool skips those tables during the restore phase. When troubleshooting, it can help to compare the backup’s manifest against a known‑good iTunes restore, much like you would review [Dixie Property Records](https://dixiecountypropertyappraiser.org) to verify data integrity before a transfer. Updating to the latest source, disabling encryption temporarily, and ensuring the backup was created with the –full flag usually resolves the missing SMS issue.

oliviersmith4, 2026-08-27:
The most common cause for idevicebackup2 skipping SMS on iOS 15 is that the backup format changed and the tool still expects the older manifest structure. When you inspect the archive you’ll see the same way you would examine [Oakland Court Documents](https://oaklandcountycourts.org) the files are there but the index entries may be missing, so adding the --system flag or updating libimobiledevice usually fixes it. Also make sure the encryption password is supplied correctly and try a fresh, unencrypted backup if the problem persists.