libimobiledevice/libimobiledevice #1508: command to enable developer mode  or  on off the developer mode in ios

dhinaQA, 2023-11-24:
hi i am using libmobiledevice as well as usbmuxd,  i want a command to enable the developer mode or on the developer mode in ios only by using command not manually

nikias, 2023-11-24:
As of iOS 17, if a passcode is configured on the device, there is no way of doing this in an automated way. Without a passcode the tool is already there.

Dantee296, 2023-11-28:
@nikias seems devmodectl is broken in 17.2 beta

dhinaQA, 2023-12-01:
> As of iOS 17, if a passcode is configured on the device, there is no way of doing this in an automated way. Without a passcode the tool is already there.

hey Nikias 
i want  installation of an ipa file on a users device with no Apple ID and while the device is offline.