libimobiledevice/libimobiledevice #1511: ipa file on a users device with no Apple ID and while the device is offline

dhinaQA, 2023-12-01:
installation of an ipa file on a users device with no Apple ID and while the device is offline.

CodeCrafterella, 2023-12-01:
I can give you a solution, you'll just need to call an API from your system and app will be installed. 
Approach me on anam.javaid@rendream.com

panderman, 2023-12-13:
Use apple MDM，user just need install a profile and then you can install an app to device via network

iteroji, 2023-12-17:
> Use apple MDM，user just need install a profile and then you can install an app to device via network

Are you sure that is correct? What you describe would require the device to be `supervised` and that requires an apple business account  and I'm pretty sure you need an appleID to be logged in on the device. Besides running an app from the app store (regardless how it was installed) requires internet connection at least once to check that the user has a license to run it. If it's an app signed with an enterprise account you can load it on the device and it should work without a connection. To accomplish it use Apple Configurator or ideviceinstaller from the cli.

AiXanadu, 2024-04-07:
You can use [https://www.ishalou.com/home], which can install applications when the device is not connected to the Internet