libimobiledevice/libimobiledevice #1515: Error compiling Cython file

orbea, 2023-12-16:
OS: `Gentoo`
libimobiledevice: https://github.com/libimobiledevice/libimobiledevice/commit/d379401ecf5c61f2417826ab14bfc9797d2c7879
libimobiledevice-glue: https://github.com/libimobiledevice/libimobiledevice-glue/commit/2ae342bdedffbcd89a0e3607bf8f2e3d2316237a
libusbmuxd: https://github.com/libimobiledevice/libusbmuxd/commit/30e678d4e76a9f4f8a550c34457dab73909bdd92
libplist: https://github.com/libimobiledevice/libplist/commit/8487d23fd2ab5683d631fd41e5f6a2f5a44d867a
cython: `3.0.5`
python: `3.11.6`

I was testing the git build and found the build doesn't succeed, but it does work when using `--without-cython`.
```
------------------------------------------------------------

imobiledevice.pyx:103:65: Cannot assign type 'void (const_idevice_event_t, void *) except * nogil' to 'idevice_event_cb_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const_idevice_event_t, void *) except * nogil'.

Error compiling Cython file:
------------------------------------------------------------
...

    cdef inline BaseError _error(self, int16_t ret):
        return NotificationProxyError(ret)

    cpdef set_notify_callback(self, object callback):
        self.handle_error(np_set_notify_callback(self._c_client, np_notify_cb, <void*>callback))
                                                                 ^
------------------------------------------------------------

notification_proxy.pxi:104:65: Cannot assign type 'void (const_char_ptr, void *) except *' to 'np_notify_cb_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const_char_ptr, void *) except *'.

Error compiling Cython file:
------------------------------------------------------------
...
        else:
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)
        if callback is None:
            err = instproxy_install(self._c_client, pkg_path, c_options, NULL, NULL)
        else:
            err = instproxy_install(self._c_client, pkg_path, c_options, instproxy_notify_cb, <void*>callback)
                                                                         ^
------------------------------------------------------------

installation_proxy.pxi:126:73: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (plist_t, plist_t, void *) except * nogil'.

Error compiling Cython file:
------------------------------------------------------------
...
        else:
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)
        if callback is None:
            err = instproxy_upgrade(self._c_client, pkg_path, c_options, NULL, NULL)
        else:
            err = instproxy_upgrade(self._c_client, pkg_path, c_options, instproxy_notify_cb, <void*>callback)
                                                                         ^
------------------------------------------------------------

installation_proxy.pxi:153:73: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (plist_t, plist_t, void *) except * nogil'.

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)

        if callback is None:
            err = instproxy_uninstall(self._c_client, appid, c_options, NULL, NULL)
        else:
            err = instproxy_uninstall(self._c_client, appid, c_options, instproxy_notify_cb, <void*>callback)
                                                                        ^
------------------------------------------------------------

installation_proxy.pxi:180:72: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (plist_t, plist_t, void *) except * nogil'.

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)

        if callback is None:
            err = instproxy_archive(self._c_client, appid, c_options, NULL, NULL)
        else:
            err = instproxy_archive(self._c_client, appid, c_options, instproxy_notify_cb, <void*>callback)
                                                                      ^
------------------------------------------------------------

installation_proxy.pxi:237:70: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (plist_t, plist_t, void *) except * nogil'.

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)

        if callback is None:
            err = instproxy_restore(self._c_client, appid, c_options, NULL, NULL)
        else:
            err = instproxy_restore(self._c_client, appid, c_options, instproxy_notify_cb, <void*>callback)
                                                                      ^
------------------------------------------------------------

installation_proxy.pxi:265:70: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (plist_t, plist_t, void *) except * nogil'.

Error compiling Cython file:
------------------------------------------------------------
...
            raise InstallationProxyError(INSTPROXY_E_INVALID_ARG)

        if callback is None:
            err = instproxy_remove_archive(self._c_client, appid, c_options, NULL, NULL)
        else:
            err = instproxy_remove_archive(self._c_client, appid, c_options, instproxy_notify_cb, <void*>callback)
                                                                             ^
------------------------------------------------------------

installation_proxy.pxi:293:77: Cannot assign type 'void (plist_t, plist_t, void *) except * nogil' to 'instproxy_status_cb_t'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (plist_t, plist_t, void *) except * nogil'.
make[2]: *** [Makefile:805: imobiledevice.c] Error 1
make[2]: Leaving directory '/tmp/libimobiledevice/cython'
make[1]: *** [Makefile:466: all-recursive] Error 1
make[1]: Leaving directory '/tmp/libimobiledevice'
make: *** [Makefile:396: all] Error 2
```
Build log: [libimobiledevice.log](https://github.com/libimobiledevice/libimobiledevice/files/13692844/libimobiledevice.log)

SourceAdFlt, 2023-12-21:
I hava the same error

SourceAdFlt, 2023-12-21:
How do you solve this?

IridiumXOR, 2024-01-11:
Same error on Fedora 39

orbea, 2024-04-23:
Should be fixed in https://github.com/libimobiledevice/libimobiledevice/commit/59ec38cb6465f794cf8e2ab204711b9a10dd9eda.

closed 2024-04-23 by orbea