libimobiledevice/libimobiledevice #1516: Error during installation of libimobiledeviceon windows

Sawrb93, 2023-12-19:
When executing ./autogen.sh final execution message is belwo

config.status: executing libtool commands

Configuration for libimobiledevice 1.3.0-202-gd379401:
-------------------------------------------

  Install prefix: .........: /mingw32
  Debug code ..............: no
  Python bindings .........: no
  SSL support backend .....: OpenSSL

  Now type 'make' to build libimobiledevice 1.3.0-202-gd379401,
  and then 'make install' for installation.

But when executing make command getting below error

$ make
make  all-recursive
make[1]: Entering directory '/home/Saurabh/libimobiledevice'
Making all in 3rd_party
make[2]: Entering directory '/home/Saurabh/libimobiledevice/3rd_party'
Making all in ed25519
make[3]: Entering directory '/home/Saurabh/libimobiledevice/3rd_party/ed25519'
  CC       add_scalar.lo
  CC       fe.lo
  CC       ge.lo
  CC       keypair.lo
  CC       key_exchange.lo
  CC       sc.lo
  CC       seed.lo
  CC       sign.lo
  CC       sha512.lo
  CC       verify.lo
  CCLD     libed25519.la
make[3]: Leaving directory '/home/Saurabh/libimobiledevice/3rd_party/ed25519'
Making all in libsrp6a-sha512
make[3]: Entering directory '/home/Saurabh/libimobiledevice/3rd_party/libsrp6a-sha512'
  CC       t_conv.lo
  CC       t_math.lo
  CC       t_misc.lo
  CC       t_truerand.lo
  CC       cstr.lo
  CC       srp.lo
  CC       srp6a_sha512_client.lo
  CC       t_sha.lo
  CCLD     libsrp6a-sha512.la
make[3]: Leaving directory '/home/Saurabh/libimobiledevice/3rd_party/libsrp6a-sha512'
make[3]: Entering directory '/home/Saurabh/libimobiledevice/3rd_party'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/Saurabh/libimobiledevice/3rd_party'
make[2]: Leaving directory '/home/Saurabh/libimobiledevice/3rd_party'
Making all in common
make[2]: Entering directory '/home/Saurabh/libimobiledevice/common'
  CC       debug.lo
  CC       userpref.lo
  CCLD     libinternalcommon.la
make[2]: Leaving directory '/home/Saurabh/libimobiledevice/common'
Making all in src
make[2]: Entering directory '/home/Saurabh/libimobiledevice/src'
  CC       idevice.lo
  CC       service.lo
  CC       property_list_service.lo
  CC       device_link_service.lo
  CC       lockdown.lo
  CC       lockdown-cu.lo
  CC       afc.lo
  CC       file_relay.lo
  CC       notification_proxy.lo
notification_proxy.c: In function 'np_notifier':
notification_proxy.c:35:18: warning: implicit declaration of function 'Sleep'; did you mean 'sleep'? [-Wimplicit-function-declaration]
   35 | #define sleep(x) Sleep(x*1000)
      |                  ^~~~~
notification_proxy.c:340:17: note: in expansion of macro 'sleep'
  340 |                 sleep(1);
      |                 ^~~~~
  CC       installation_proxy.lo
  CC       sbservices.lo
  CC       mobile_image_mounter.lo
  CC       screenshotr.lo
  CC       mobilesync.lo
  CC       mobilebackup.lo
  CC       house_arrest.lo
  CC       mobilebackup2.lo
  CC       misagent.lo
  CC       restore.lo
  CC       diagnostics_relay.lo
  CC       heartbeat.lo
  CC       debugserver.lo
  CC       webinspector.lo
  CC       mobileactivation.lo
  CC       preboard.lo
  CC       companion_proxy.lo
  CC       reverse_proxy.lo
  CC       syslog_relay.lo
  CC       bt_packet_logger.lo
  CCLD     libimobiledevice-1.0.la
C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/13.2.0/../../../../i686-w64-mingw32/bin/ld.exe: warning: resolving _Sleep by linking to _Sleep@4
Use --enable-stdcall-fixup to disable these warnings
Use --disable-stdcall-fixup to disable these fixups
`_Sleep' referenced in section `.text' of .libs/notification_proxy.o: defined in discarded section `.text' of C:/msys64/mingw32/lib/libkernel32.a(libkernel32s01416.o)
collect2.exe: error: ld returned 1 exit status
make[2]: *** [Makefile:552: libimobiledevice-1.0.la] Error 1
make[2]: Leaving directory '/home/Saurabh/libimobiledevice/src'
make[1]: *** [Makefile:466: all-recursive] Error 1
make[1]: Leaving directory '/home/Saurabh/libimobiledevice'
make: *** [Makefile:396: all] Error 2

mexmer, 2023-12-19:
most likely there is missing reference to <windows.h>
but adding on line 35 (eg. right befor sleep definition)
```
#include <synchapi.h>
```
should be enough (must be inside #ifdef WIN32)

Sawrb93, 2023-12-20:
Thanks @mexmer 
But I am not familiar with the code can you please tell me in which file I have to add 
#include <synchapi.h>

I tried on 
**tools/idevicedevmodectl.c**
 file but it did'nt work

mexmer, 2023-12-20:
error you got was in notification_proxy.c
so you put it here
https://github.com/libimobiledevice/libimobiledevice/blob/d379401ecf5c61f2417826ab14bfc9797d2c7879/src/notification_proxy.c#L35

idevmodectl.c includes whole windows.h, so it's not needed there
including either **windows.h** or **synchapi.h** should work, because **windows.h** includes **synchapi.h**, i prefer going with minimal requirements
but i compile under msvc builder, in cygwin environment using
```
#include <windows.h>
```
is better choice, because _syncapi.h_ is not present in cygwin (it's part of windows SDK)

Sawrb93, 2023-12-21:
It worked.
Thanks @mexmer really appreciated

closed 2023-12-21 by Sawrb93