libimobiledevice/libimobiledevice #1518: ERROR: Could not connect to lockdownd: SSL error (-5)

gotname, 2023-12-27:
I am encountering this SSL error -5 issue, and the situation is as follows:
Occasionally, it takes two clicks on "Trust" to establish a connection.
After clicking "Trust" for the first time, an SSL error (-5) occurs.
I need to unplug the phone, plug it in again, and click "Trust" again for everything to work normally.
There is approximately a 40% chance of encountering the SSL error after the first "Trust" click.

When the error occurs, I checked the pairing list, and the device is listed !
Operating Environment: Windows
Versions Used: 1.3.0-174-g9a5e78f, 1.2 - both versions encounter this issue
Tested Devices:
iPhone 7 iOS 13.3.1 (Activated, password-protected, unlocked)
iPhone 11 iOS 15.4 (Activated, no password protection, unlocked)
I believe this issue might occur with other devices or systems.
SSL Version: 1.1.0e
Both "pair" and "unpair" commands fail when the error occurs.
Screenshots of cmd commands in Windows are provided below. Does anyone know where the issue might be, and how to fix it?

[image]

mexmer, 2024-01-01:
on windows libimobiledevice tools use applemobiledeviceservice that is shipped with itunes update/reinstall itunes and check if issue still persists.

gotname, 2024-01-02:
@mexmer 
I have tried updating/reinstalling iTunes, but the issue still persists. However, I've noticed that when the SSL error (-5) occurs, doing nothing and simply launching the third-party app 3uTools resolves the problem, and the phone being reported with an error is automatically recognized as normal.

mexmer, 2024-01-02:
3utools internally use libimobilevice. maybe there is issue with openSSL version. we are using 1.1.1q and never observed this issue.

AiXanadu, 2024-01-26:
My solution is to use a third-party process (https://github.com/libxcc/xcc-rundll) to call the connection function and time out for 5 seconds.
Then wait for it to end in the main program. If it connects successfully, we will connect normally. If it times out for 5 seconds without connecting. Then an SSL error is returned, and we wait (about 3 seconds) to start the process again to connect.
This ensures that when we actually connect, we won't get an SSL error, which can take a long time to return.