libimobiledevice/libimobiledevice #1520: why there is no video tutorial hot to install all of that, a lot of people doesn't know a thing about commands etc

httpantek, 2024-01-03:


jzhang1043, 2024-01-05:
I'm just embarrassed for you when I see questions like this. If you don't know how to do derivative becasue you don't know what is 1 + 1, then you should learn 1 + 1 first. No one in a college is going to make a video tutorial for you to learn 1 + 1 when you don't know how to do a derivative. You should learn 1 + 1 outside of college if you don't know. 

So...people like you should watch some tutorials about commands etc before even taking a glance at this project, and your question makes no sense because all the necessary installation instructions is provided in the readme.

steam3d, 2024-01-10:
Each link contains `Installation / Getting started`.

1. [libplist](https://github.com/libimobiledevice/libplist)
2. [libusbmuxd](https://github.com/libimobiledevice/libusbmuxd)
3. [libimobiledevice-glue](https://github.com/libimobiledevice/libimobiledevice-glue)

4. [To install cython use:](https://cython.readthedocs.io/en/latest/src/quickstart/install.html)

```
sudo apt-get install build-essential python3-dev
```

5. [libimobiledevice](https://github.com/libimobiledevice/libimobiledevice)

6. [After install](https://github.com/libimobiledevice/libimobiledevice/issues/1081)

```
sudo ldconfig
```