libimobiledevice/libimobiledevice #1522: ./utils.h:55:2: error: redefinition of enumerator 'PLIST_FORMAT_XML'

PJStation, 2024-01-15:
libimobiledevice1.3.0 and libplist2.3.0 redefinition of enumerator 'PLIST_FORMAT_XML'
`plist.h:145:9: note: previous declaration is here
        PLIST_FORMAT_XML     = 1,  /**< XML format */
        ^
In file included from userpref.c:69:
./utils.h:55:2: error: redefinition of enumerator 'PLIST_FORMAT_XML'
        PLIST_FORMAT_XML,`

mexmer, 2024-01-15:
i take it, you trying to compile source from tagged versions.
mind plist is from 2023, while libimobiledevice tag you are compiliing is over 4 years old, there was done refactoring in libimobile device, so if you want to compile libimobiledevice, use latest source code, not 4 years old tag. also you will need to compile 
[libimobiledevice-glue](https://github.com/libimobiledevice/libimobiledevice-glue)

PJStation, 2024-01-15:
I use [libimobiledevice-glue](https://github.com/libimobiledevice/libimobiledevice-glue) git 1.0.0

PJStation, 2024-01-15:
> i take it, you trying to compile source from tagged versions. mind plist is from 2023, while libimobiledevice tag you are compiliing is over 4 years old, there was done refactoring in libimobile [...]

I use [libimobiledevice-glue](https://github.com/libimobiledevice/libimobiledevice-glue) git 1.0.0

mexmer, 2024-01-15:
> I use [libimobiledevice-glue](https://github.com/libimobiledevice/libimobiledevice-glue) git 1.0.0

yes, but you have error in libmobiledevice 1.3.0, tag for that version is from [2020-06-15](https://github.com/libimobiledevice/libimobiledevice/releases/tag/1.3.0)

SunPengJu-MX, 2024-01-16:
libmobiledevice master is ok

PJStation, 2024-01-16:
switch libmobiledevice master is ok

closed 2024-01-16 by PJStation