libimobiledevice/libimobiledevice #1531: possible to enter recovery from a pairing-restricted device?

ianmmyers, 2024-02-08:
I'm working on making a process where our corporate-issued iPads can be wiped if the user forgets the passcode and we can't reset it remotely. These are managed in an MDM, and device pairing is prevented by a profile.

Given all of this, is it possible to use  `ideviceenterrecovery` to put a device into recovery mode, or do we have to rely on the user using the key combo on the device itself?

nikias, 2024-02-09:
It cannot be put into recovery automatically without pairing. If the MDM doesn't allow removal of the profile I don't think there is a way. However there is a feature I through MDM I believe that would allow to reset the passcode. But I am not 100% sure.

mexmer, 2024-02-09:
if your devices are managed with mdm, you should have also ability to wipe or unlock device remotely, since you block device pairing, it's only way to do it.

ianmmyers, 2024-02-09:
Thanks all. So we can erase the passcode via MDM, that works as expected. The issue I'm trying to solve for is the case where a user has both forgotten a passcode and they don't have WiFi. As this answers my question I'm closing this

closed 2024-02-09 by ianmmyers

ianmmyers, 2024-02-09:
OK, so there actually **is** a solution to this. The trick is in your MDM, in the Restrictions payload, you have to set `allowUnpairedExternalBootToRecovery` to `TRUE`. With this you can just do

`ideviceenterrecovery <UUID>`

and you will enter Recovery as expected

[link to Apple doc](https://developer.apple.com/documentation/devicemanagement/restrictions)