libimobiledevice/libimobiledevice #1533: Get error druing install the libimobiledevice on windows

thanhphong97, 2024-02-24:
I use the  Msys2/MINGW64 to combine the source that clones from git, but when I enter "make" to build libimobiledevice I get the message

`
$ make
make  all-recursive
make[1]: Entering directory '/usr/src/libimobiledevice'
Making all in 3rd_party
make[2]: Entering directory '/usr/src/libimobiledevice/3rd_party'
Making all in ed25519
make[3]: Entering directory '/usr/src/libimobiledevice/3rd_party/ed25519'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/usr/src/libimobiledevice/3rd_party/ed25519'
Making all in libsrp6a-sha512
make[3]: Entering directory '/usr/src/libimobiledevice/3rd_party/libsrp6a-sha512'
  CC       t_misc.lo
In file included from C:/msys64/usr/include/sys/fcntl.h:3,
                 from C:/msys64/usr/include/fcntl.h:12,
                 from t_misc.c:39:
C:/msys64/usr/include/sys/_default_fcntl.h:187:8: error: redefinition of 'struct flock'
  187 | struct flock {
      |        ^~~~~
In file included from C:/msys64/usr/include/sys/types.h:222,
                 from t_defines.h:91,
                 from t_misc.c:30:
C:/msys64/usr/include/machine/types.h:35:8: note: originally defined here
   35 | struct flock {
      |        ^~~~~
In file included from C:/msys64/usr/include/openssl/e_os2.h:234,
                 from C:/msys64/usr/include/openssl/err.h:21,
                 from t_sha.h:41,
                 from t_misc.c:46:
C:/msys64/usr/include/inttypes.h:28: warning: "__STRINGIFY" redefined
   28 | #define __STRINGIFY(a) #a
      |
In file included from C:/msys64/mingw64/include/_mingw.h:10,
                 from C:/msys64/mingw64/include/corecrt.h:10,
                 from C:/msys64/mingw64/include/crtdefs.h:10,
                 from C:/msys64/mingw64/include/stddef.h:7,
                 from C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.2.0/include/stddef.h:1,
                 from C:/msys64/usr/include/stdlib.h:16,
                 from t_defines.h:78:
C:/msys64/mingw64/include/_mingw_mac.h:10: note: this is the location of the previous definition
   10 | #define __STRINGIFY(x) #x
      |
In file included from C:/msys64/usr/include/errno.h:9,
                 from C:/msys64/usr/include/openssl/err.h:46:
C:/msys64/usr/include/sys/errno.h:18: warning: "errno" redefined
   18 | #define errno (*__errno())
      |
C:/msys64/mingw64/include/stddef.h:19: note: this is the location of the previous definition
   19 | #define errno (*_errno())
      |
make[3]: *** [Makefile:433: t_misc.lo] Error 1
make[3]: Leaving directory '/usr/src/libimobiledevice/3rd_party/libsrp6a-sha512'
make[2]: *** [Makefile:414: all-recursive] Error 1
make[2]: Leaving directory '/usr/src/libimobiledevice/3rd_party'
make[1]: *** [Makefile:468: all-recursive] Error 1
make[1]: Leaving directory '/usr/src/libimobiledevice'
make: *** [Makefile:398: all] Error 2
`

Does someone have an idea?

HTWang-CHN, 2024-04-12:
Hi, brother, I have the same problem, have you solved it?

diamond261, 2024-05-21:
windows cannot use libimobiledevice only linux can do it