libimobiledevice/libimobiledevice #1540: idevicebackup2 issue

devloopsmilan, 2024-03-01:
----------------------------------
Versions info
----------------------------------
OS : Windows 11 (22H2)
IPhone versione 6 (IOS 12.5.7)
C:\libimobiledevice-win64>idevice_id --version
idevice_id 1.0.0

C:\libimobiledevice-win64>idevicepair -v
idevicepair 1.0.0

C:\libimobiledevice-win64>idevicebackup2 -v
idevicebackup2 1.0.0

----------------------------------
Commands executed
----------------------------------

C:\libimobiledevice-win64>idevice_id
d5d19042dfb2b441722fbf334084e7d4ca63afef (Network)

(I have successfully executed unpair and pair of the iphone)
C:\libimobiledevice-win64>idevicepair list
d5d19042dfb2b441722fbf334084e7d4ca63afef

Now, i try to execute network backup (here i have the issue. It seems to me that a loop begins. i need to do CTRL+C to stop IT) : 

- This is the NO DEBUG command output (it stops here "Negotiated Protocol Version 2.1") : 
C:\libimobiledevice-win64>idevicebackup2 backup --full c:\bck -u d5d19042dfb2b441722fbf334084e7d4ca63afef -n
Backup directory is "c:\bck"
Started "com.apple.mobilebackup2" service on port 52018.
Negotiated Protocol Version 2.1

- This is the DEBUG command outputVersion: 

C:\libimobiledevice-win64>idevicebackup2 backup --full c:\bck -u d5d19042dfb2b441722fbf334084e7d4ca63afef -n -d
Backup directory is "c:\bck"
11:21:19 D:\a\1\s\libimobiledevice\src\lockdown.c:650 lockdownd_client_new(): device udid: d5d19042dfb2b441722fbf334084e7d4ca63afef
11:21:19 D:\a\1\s\libimobiledevice\src\lockdown.c:379 lockdownd_query_type(): called
11:21:19 D:\a\1\s\libimobiledevice\src\property_list_service.c:132 internal_plist_send(): sending 287 bytes
11:21:19 D:\a\1\s\libimobiledevice\src\service.c:144 service_send(): sending 4 bytes
11:21:19 D:\a\1\s\libimobiledevice\src\idevice.c:548 idevice_connection_send(): internal_connection_send 4, sent 4
11:21:19 D:\a\1\s\libimobiledevice\src\service.c:144 service_send(): sending 287 bytes
11:21:19 D:\a\1\s\libimobiledevice\src\idevice.c:548 idevice_connection_send(): internal_connection_send 287, sent 287
11:21:19 D:\a\1\s\libimobiledevice\src\property_list_service.c:137 internal_plist_send(): sent 287 bytes
11:21:19 D:\a\1\s\libimobiledevice\src\property_list_service.c:138 internal_plist_send(): printing 287 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>idevicebackup2</string>
        <key>Request</key>
        <string>QueryType</string>
</dict>
</plist>
11:21:19 D:\a\1\s\libimobiledevice\src\property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
11:21:19 D:\a\1\s\libimobiledevice\src\property_list_service.c:211 internal_plist_receive_timeout(): 297 bytes following
11:21:19 D:\a\1\s\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 297 bytes
11:21:19 D:\a\1\s\libimobiledevice\src\property_list_service.c:253 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Request</key>
        <string>QueryType</string>
        <key>Type</key>
        <string>com.apple.mobile.lockdown</string>
</dict>
</plist>
[... 649 lines cut ...]
<dict>
        <key>Command</key>
        <string>Shutdown</string>
</dict>
</plist>
11:21:23 D:\a\1\s\libimobiledevice\src\idevice.c:652 idevice_connection_receive_timeout(): SSL_read 4, received 4
11:21:23 D:\a\1\s\libimobiledevice\src\property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
11:21:23 D:\a\1\s\libimobiledevice\src\property_list_service.c:211 internal_plist_receive_timeout(): 237 bytes following
11:21:23 D:\a\1\s\libimobiledevice\src\idevice.c:652 idevice_connection_receive_timeout(): SSL_read 237, received 0
11:21:23 D:\a\1\s\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 0 bytes
11:21:23 D:\a\1\s\libimobiledevice\src\idevice.c:652 idevice_connection_receive_timeout(): SSL_read 237, received 0
11:21:23 D:\a\1\s\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 0 bytes
11:21:23 D:\a\1\s\libimobiledevice\src\idevice.c:652 idevice_connection_receive_timeout(): SSL_read 237, received 0
11:21:23 D:\a\1\s\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 0 bytes
11:21:23 D:\a\1\s\libimobiledevice\src\idevice.c:652 idevice_connection_receive_timeout(): SSL_read 237, received 0
11:21:23 D:\a\1\s\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 0 bytes
11:21:23 D:\a\1\s\libimobiledevice\src\idevice.c:652 idevice_connection_receive_timeout(): SSL_read 237, received 0
11:21:23 D:\a\1\s\libimobiledevice\src\property_list_service.c:224 internal_plist_receive_timeout(): received 0 bytes
^C

----------------------------------
Expected
----------------------------------
Backup created in C:\BCK
The backup is not created and the folder remains empty

devloopsmilan, 2024-03-08:
Hi,
Any news on this?
I have checked :

1. Wi-Fi flag in iTunes is checked
2. Device is successfull Paired
3. Host and phone are on the same network
4. Windows firewall is disabled
5. The same happen also with the last master repo version (i follower this steps https://github.com/libimobiledevice/libimobiledevice/issues/582 to compile it on Windows)

Can you confirm that anyone, creating successfully a backup via Wi-Fi works under Windows? and possibly provide additional  about the process? Thank you very much for any suggestions that may arrive

ekscrypto, 2024-03-11:
The tool is deprecated. It expects a compatibility version of 300 and most recent phones have compatibility level 400.  We have the same issue on macOS when using it.