libimobiledevice/libimobiledevice #1544: idevicebackup2 incremental backups fail because snapshot directory is not empty

niels-beier, 2024-03-23:
When I perform a backup using `idevicebackup2` of my iPad (Air 5), in the end when removing files the snapshot directory can't be deleted since it's not empty, i.e. `Could not remove './UDID_OF_MY_DEVICE/Snapshot': Directory not empty (39)`.
I already found a workaround by deleting the old backup and doing a new full one, but in the past there was no problem at all with incremental backups. I tried this a few times now and all incremental backups fail in the end.

mexmer, 2024-03-26:
Snapshot folder remains, when backup fails. From my experience, if you do same thing on windows, when doing backup trough Apple iTunes application same thing can happen, if you for example disconnect phone during backup. then you get funny error in desktop application, you need to search on their website for help.

probly best course of action will be, if idevicebackup app search for "snapshot" folder prior sending request to device and delete it's contents to prevent this error. other possibility will be to delete folder after failed backup.

mind whole backup process is controlled by device, not by idevicebackup2, it only connects to backup service inside phone, and phone tells it, what to do.

niels-beier, 2024-03-30:
> other possibility will be to delete folder after failed backup.
This doesn't help, since when removing the folder and retrying the backup the same error occurs again. The device is also connected at all times, I also used all cables I got to prevent some of them being broken.

mexmer, 2024-04-02:
> > other possibility will be to delete folder after failed backup.
> 
> This doesn't help, since when removing the folder and retrying the backup the same error occurs again. The device is also connected at all times, I also used all cables I got to prevent some of them being broken.

did your previous backup ended with success, if you do full backup and it succeed, there should be no Snapshot folder inside your device backup folder, and incremental backup afterwards should succeed.

as for test.
do backup to some new folder (you can actually set backup root on your disk on idevicebackup2 commandline). open backup folder (targetpath/UDID) and check, if there is Snapshot folder present. if Snapshot folder is present, then your backup didn't finish properly.

if you go by timeline of device backup
computer ask device to do backup, and sends target location on disk, where it should be placed
device asks about content of targetfolder/udid, device asks about free space
device sends status.plist (where is write current backup status)
device then ask computer to create targetfolder/udid/SnapShot folder
device then ask for creating subsfolders, and sends files that are backed up
device creates manifest database, and send it to computer
device asks computer to move content of shapshot folder to main folder
devices asks computer to delete snapshot folder (which should be empty at that time)
device sends manifest.plist
device sends again status.plist with status finished

0xshuzo, 2026-10-10:
Just to bring this issue to an end: updating the device helped at the time, so if someone has the same problem, updating the device (if possible) may fix the issue.

closed 2026-10-10 by 0xshuzo