libimobiledevice/libimobiledevice #1545: !!! mobilegestalt deprecated for ios 17.4, all the querries ware no working

jsonlee1688, 2024-03-27:
```
- (NSString *)readDeviceRegulatoryModelNumber
{
    @autoreleasepool {
        NTBTask *task = [[NTBTask alloc] initWithLaunchPath:KCUtility.KC_IDEVICEDIAGNOSTICS_PATH];
        task.arguments = @[@"-u",self.deviceUDID,@"mobilegestalt",@"97JDvERpVwO+GHtthIh7hA"];
        NSString *opt = [task waitForOutputString];
        NSLog(@"%s opt:%@ deviceUDID:%@",__func__,opt,self.deviceUDID);
        NSLog(@"readDeviceRegulatoryModelNumber Thread:%@",[NSThread currentThread]);
        id regulatoryModelNumber = [self parsexmlstring:opt forKey:@"97JDvERpVwO+GHtthIh7hA"];
        return regulatoryModelNumber?regulatoryModelNumber:@"";
    }
}
```

I was reading RegulatoryModelNumber just like "A1234", but it returned nothing since upgrade to iOS 17.4.  What happend with iOS 17.4? How to fix? Can any one help?

jsonlee1688, 2024-03-29:
is there any body help?

ekscrypto, 2024-04-03:
+1, curious about this

AiXanadu, 2024-04-07:
Also looking forward to new progress

hoangnam714, 2024-06-25:
+1, curious about this