libimobiledevice/libimobiledevice #1546: Could not start service com.apple.mobilebackup2.

Lautarob1, 2024-03-27:
Error using idevicebackup2 backup --full --password 'xxxx': Could not start service com.apple.mobilebackup2.

Is this caused for something I should have installed and I didn't or is a problem caused by an Apple recent change?

Running a MacBook Pro (Intel), macOS Sonoma and the connected iPhone is 13 Pro Max running iOS 17.3

Thanks!

nikias, 2024-03-27:
Seems like you are using an old version, newer code prints an error description of why it doesn't work. Run with `-d` and see if it prints anything? But I am suggesting to build the latest code.

Lautarob1, 2024-03-28:
Thanks Nikias, 

The version I used is 1.3, which I believe is the current one. Furthermore, the Mac where I tested did not have any previous version. As to use -d, I will try later today but I don't expect much of it, as the message is clear:  com.apple.mobilebackup2 did not start.... My question still is if didn't start because some I did or failed to do, or if it is being caused by an Apple issue with libimobile.

Lautarob1, 2024-03-28:
Nikias, after running the command with -d, I got lots of info (400 lines or so on the terminal). At first, it seems to be ok:

15:29:05 lockdown.c:1168 lockdownd_start_session(): Session startup OK
15:29:05 lockdown.c:1180 lockdownd_start_session(): SessionID: B06B9E33-6FFA-4680-99C5-6ADA46CC6025
15:29:05 lockdown.c:1187 lockdownd_start_session(): Enable SSL Session: true
15:29:05 idevice.c:1077 idevice_connection_enable_ssl(): Performing SSL handshake
15:29:05 idevice.c:1101 idevice_connection_enable_ssl(): SSL mode enabled, TLSv1.2, cipher: ECDHE-RSA-AES256-GCM-SHA384

But then this happened:

15:29:05 lockdown.c:1239 lockdownd_build_start_service_request(): ERROR: Failed to retrieve the escrow bag from the device's record
15:29:05 lockdown.c:262 lockdownd_stop_session(): stopping session B06B9E33-6FFA-4680-99C5-6ADA46CC6025

15:29:05 lockdown.c:278 lockdownd_stop_session(): success
15:29:05 idevice.c:1193 idevice_connection_disable_bypass_ssl(): Skipping bidirectional SSL shutdown. SSL error code: 5

15:29:05 idevice.c:1208 idevice_connection_disable_bypass_ssl(): SSL mode disabled
ERROR: Could not start service com.apple.mobilebackup2.

And ended as:

15:29:05 notification_proxy.c:299 np_get_notification(): NotificationProxy died!
15:29:05 notification_proxy.c:62 np_unlock(): Unlocked
15:29:05 idevice.c:1193 idevice_connection_disable_bypass_ssl(): Skipping bidirectional SSL shutdown. SSL error code: 5

15:29:05 idevice.c:1208 idevice_connection_disable_bypass_ssl(): SSL mode disabled

Any help will be highly appreciated,

Thanks!

mexmer, 2024-04-02:
> Thanks Nikias,
>
> The version I used is 1.3, which I believe is the current one. Furthermore, the Mac where I tested did not have any previous version. As to use -d, I will try later today but I don't [...]

current version is HEADE from master branch, there is no tag for version that is working with newer ioses.

chriseturi, 2026-08-17:
Does updating to the new version 1.4 fix this issue?

mexmer, 2026-08-18:
code was updated since then. so it should work. i always use HEAD just in case.