libimobiledevice/libimobiledevice #1548: Unable to restore from backup

amcooper, 2024-03-29:
Attempts to restore from my iPhone backup are failing. Basic info and steps taken follow below.

#### Basic info
- Arch Linux; kernel 6.8.1
- iOS 17.4.1
- libimobiledevice 1.3.0-190-g6fc41f5

#### Steps
- Plug the iPhone into the laptop USB port.
- Execute `idevicebackup2 restore $BACKUP_DIR`
- Progress proceeds to 100%
- iPhone reboots & goes through setup steps as though new. When it asks where to transfer data from, I choose "no transfer" \[?\] (as none of the other options makes any more sense)
- Backup data is not actually restored

Any ideas?

datatags, 2024-04-17:
Also having this issue, I tried adding the arguments `--system --settings` which restores photos, app layout, wallpaper... but third-party apps and some Apple app data aren't restored. I can collect logs and such if it would help diagnose.

C0D3-M4513R, 2024-09-21:
I have similar issues.
Even when I try restoring without the initial setup running, my apps are not visible, my wallpaper is wrong and my data is not restored.

martinvonwittich, 2025-03-27:
Same here. `idevicebackup2 restore --settings --source <old UDID> -i .ios-backup` completed with the following message:

```
The device should reboot now.
Restore Successful.
```

But after rebooting I get the "hello" screen where I'm forced to set it up with "no transfer", and apparently nothing has been restored.

Alyer7777, 2026-02-01:
Hello there

I'm attempting to restore an ios 26.2.1 backup from an iphone 14 to an iphone 16... Unfortunately, even if Info.plist is into the correct directory, the terminal persist to says its not there : 

alyer@alyer-Legion-Y540-17IRH:~$ 

idevicebackup2 --source [udid] --interactive restore --system --reboot --copy --settings /home/alyer/Documents/iphone-backup/[udid]/

ERROR: Backup directory "/home/alyer/Documents/iphone-backup/[udid]/" is invalid. No Info.plist found for UDID [udid].

I tried to set the commands in any order but do not works

Alyer7777, 2026-02-01:
@datatags oh, that would explain that it doesn't find the files, I'll try that 
Thanks for your reply 😁