libimobiledevice/libimobiledevice #1549: Can't install the library

Blackspell01, 2024-03-29:
I followed the official instructions and typed 
```
apt-get install \
	build-essential \
	pkg-config \
	checkinstall \
	git \
	autoconf \
	automake \
	libtool-bin \
	libplist-dev \
	libusbmuxd-dev \
	libimobiledevice-glue-dev \
	libssl-dev \
	usbmuxd
```
which results in `E: unable to locate package libimobiledevice-glue-dev`

after procceeding with cloning the repo it fails again: 
```
configure: error: Package requirements (libplist-2.0 >= 2.3.0) were not met:
Requested 'libplist-2.0 > 2.3.0' but version of libplist is 2.2.0
```
I don't know what im doing wrong. My actual goal is to perform backups wirelessly to my host machine.

Eventually I was successful after following another tutorial. They suggested to use this command
 `apt install -y usbmuxd libimobiledevice6 libimobiledevice-utils libavahi-compat-libdnssd-dev curl wget`

Afterwards I could pair my iPhone 15. 

Yet I haven't understood how to do the backup wirelessly as advertised.

What am I missing?

iFred09, 2024-04-10:
Hello,

You can compile and install the latest version of libplist which works with libimobiledevice here: https://github.com/libimobiledevice/libplist

After, you can compile libusbmuxd, libimobiledevice-glue and libimobiledevice (in that order) and it will works. These are available on https://github.com/libimobiledevice/

Blackspell01, 2024-04-10:
Ok thanks, this seemed to work. Though, I still don't know how to backup wirelessly... Can you help me with that?

iFred09, 2024-04-11:
I think that you can't do this wirelessly with libimobiledevice...plug your device to your computer and it will be fine!

Blackspell01, 2024-04-11:
My server is not located comfortably enough to plug in my iPhone every time. Goal is also to backup from somewhere else. Otherwise I could just use a usb ssd and plug it into my Mac. 
I was just wondering because on the website it is advertised to work wirelessly.

mexmer, 2024-04-12:
wireless backup is only possible to computer that is located on same network as phone, that is limitation imposed by Apple (Eg. built into operating system). if you want to upload your backup to different site/network, you would need first to setup machine, that will your phone backup to (at same network as phone), and then that computer will need to upload it to your offsite target, after backup is finished.

only direct offsite backup of iphone is to icloud.

there is exists tools that can do limited backup (mostly focusing on doing offsite backup of photos, sometimes also contacts, but no application can backup other applications data).

mexmer, 2024-04-15:
apple designed it that way, nothing to do on limbimobile side. iOS device advertises it presence on local network, establishes "trust" with computer on that local network, and only to that computer does backup (again on local network), discovery protocols like bonjour or SSDP are designed to work only on local network.

technically, you can setup site to site VPN, and allow forwarding discovery and broadcast packets to other end of VPN, but you will need to setup network for that. (eg. networks on both sides of VPN need to be part of same network). again it's stuff out of scope of libimobile.

"network" doesn't equal internet.

Blackspell01, 2024-04-15:
Ok, thanks for the clarification. Im gonna use iMazing then

mexmer, 2024-04-15:
iMazing utilizes libimobile, so if they doing what you want, it's because they programmed that additional functionality, that is required to do that. 

there are multiple software that use libimobile and extend it's functionality, which in my opinion is great.

Blackspell01, 2024-04-15:
its actually pretty simple, I want to backup my iPhone from time to time but not waste any space on my MacBook so the files should be stored on my nas. 
[Over there](https://github.com/libimobiledevice/libusbmuxd/issues/88) my wish is described but I don't understand every step so I might just give up at this point

codebar33, 2024-04-15:
A quick note for @mexmer: As the developers of iMazing, we don't utilize libimobiledevice. Instead, iMazing directly links the Apple MobileDevice framework/dll :)

mexmer, 2024-04-15:
> A quick note for @mexmer: As the developers of iMazing, we don't utilize libimobiledevice. Instead, iMazing directly links the Apple MobileDevice framework/dll :)

Really, i though i've seen libimobile in some version we had tested in past.
must have mistaken it with some other tool then.

Thanks for clarification, and appologize for my misinformation.

codebar33, 2024-04-15:
No worries, we have always relied on Apple's MobileDevice library. This doesn't make much of a difference compared to using libimobiledevice, but it typically results in fewer breaking changes, especially for updating or restoring iOS. I'm always open to discussing various subjects related to MobileDevice :)

closed 2025-01-25 by Blackspell01