libimobiledevice/libimobiledevice #1553: ERROR: Could not start the simulatelocation service. Make sure a developer disk image is mounted!

basvdijk, 2024-04-12:
When running `idevicesetlocation` on MacOS 14.4.1 with iOS 17.4.1 I get the error:

```bash
ERROR: Could not start the simulatelocation service. Make sure a developer disk image is mounted!
```

I am wondering if some got it to work and what steps are needed. Or is with the new implementation of the images, `idevicesetlocation` not going to work anymore?

This is issue could be related to https://github.com/libimobiledevice/libimobiledevice/issues/1547

mexmer, 2024-04-12:
ideviceimagemounter doesn't support iOS17, apple changed how developer images work in ios 17, so you will need first use xcode, to upload developer image into your phone, and they you will be able to play with idevicesetlocation.

basvdijk, 2024-04-12:
@mexmer Are there any instructions on how to upload the developer image to my phone?

mexmer, 2024-04-12:
as i said, use Xcode on mac. there is no support for uploading ios17 images in libimobile project atm.
xcode will automatically create and load personal developer image into your ios device, when you try debugging application.
other than that. i don't know about any way.

there is python project based on libimobile
https://github.com/doronz88/pymobiledevice3
but i have no idea, if it can generate personalized developer image (new format, that apple decided to use since ios17).

basvdijk, 2024-04-12:
@mexmer Maybe I misunderstood, just to be sure: Do you mean with "use XCode on Mac":
- A: Use XCode to debug whatever you did with `libimobiledevice` like fake a GPS location
- B: Use XCode to upload the developer image, then you can use `libimobiledevice` to fake a GPS location.

I tried `[pymobiledevice3](https://github.com/doronz88/pymobiledevice3)` and was able to fake GPS locations. It is more work and involves quite some work though.

mexmer, 2024-04-12:
yes i did mean use xcode on mac, then you can do B ... but if you can do it with that python project, i guess you don't need xcode then.

basvdijk, 2024-04-12:
@mexmer When I use option B and run my app on iOS, I assume XCode has uploaded the developer image. However I still get:
`ERROR: Could not start the simulatelocation service. Make sure a developer disk image is mounted!`

Do you know a way to verify or force XCode to upload the developer image?

mexmer, 2024-04-15:
xcode uploads developer image to device when you debug application. i had not tested if it works with simulatelocation, since my application doesn't use it. tbh. with ios 17.4 even idevicedebug doesn't work for me, hence why i'm using xcode.

basvdijk, 2024-04-16:
That is exactly my problem. When I debug via XCode, still idevicedebug doesn't work. I ended up using https://github.com/doronz88/pymobiledevice3

lingshigang, 2024-05-08:
pymobiledevice3 developer dvt simulate-location set -- 40.690008 -174.045843
loper command
Usage: pymobiledevice3 developer dvt simulate-location set
           [OPTIONS] LATITUDE LONGITUDE
Try 'pymobiledevice3 developer dvt simulate-location set -h' for help.

Error: Got unexpected extra arguments (--tunnel 00008101-000C6DEA01C1034E)

lingshigang, 2024-05-09:
[pymobiledevice3](https://github.com/doronz88/pymobiledevice3)  window Could not start the simulatelocation service

KovtonyckIlya, 2024-07-10:
any updates here?

xclidongbo, 2024-07-24:
any updates here?

xclidongbo, 2024-07-24:
Failed to start service. Possible reasons are:
- If you were trying to access a developer service (developer subcommand):
    - Make sure the DeveloperDiskImage/PersonalizedImage is mounted via:
      > python3 -m pymobiledevice3 mounter auto-mount

    - If your device iOS version >= 17.0:
        - Make sure you passed the --rsd option to the subcommand
          https://github.com/doronz88/pymobiledevice3#working-with-developer-tools-ios--170

- Apple removed this service

- A bug. Please file a bug report:
  https://github.com/doronz88/pymobiledevice3/issues/new?assignees=&labels=&projects=&template=bug_report.md&title=