libimobiledevice/libimobiledevice #1560: master fails to build with libressl 3.9.1

Un1q32, 2024-04-22:
The lastest git commit uses functions that were removed in libressl 3.9.1 resulting in a failed build

nikias, 2024-04-23:
Would be helpful to get some output so I can work around the functions that have been removed

orbea, 2024-04-23:
I didn't notice this issue before now, but this should fix at least some of the issues?

https://github.com/libimobiledevice/libimobiledevice/pull/1561
https://github.com/libimobiledevice/libimobiledevice/pull/1562

Un1q32, 2024-04-23:
> I didn't notice this issue before now, but this should fix at least some of the issues?
> 
> #1561 #1562

Those are the functions I had issues with, I'll try to build with those patches

Un1q32, 2024-04-23:
The build works with those 2 patches

orbea, 2024-04-27:
@Un1q32 This should be fixed now in the master branch?

Un1q32, 2024-04-27:
yes, master builds now

closed 2024-04-27 by Un1q32