libimobiledevice/libimobiledevice #1565: files coming off of device twice

blakeb211, 2024-05-04:
I noticed that when using the library (c/c++ interface), files with the same exact name (coming from **mb2_receive_filename**) and path come off of the device **twice.**

Is this normal and known behavior? 

Is there a reasoning behind it from the iphone side?

Most importantly - is there a way to tell that you are receiving a file that will come again later at the the time of it coming for the first time? It seems like the first one is a partial and the second one is always the complete and correct one, but I have not fully confirmed that.

Please and thank you!! Great library!

nikias, 2024-05-04:
What's the filename?

blakeb211, 2024-05-05:
Here is one...a sqlite file. Many of them are sqlite files. I'm not certain that all of them are though but its possible.  `64d0019cb3d46bfc8cce545a8ba54b93e7ea9347`

nikias, 2024-05-05:
The device is controlling the entire process. Files are re-transmitted when the file to back up changed during the backup, for example.