libimobiledevice/libimobiledevice #1571: ideviceinfo shows wrong productType

Lohithlee, 2024-06-13:
i currently have iPhone13, when i give 'ideviceinfo  | grep  'ProductType'' command it returns as iPhone14 and same goes with other models
it always returns one model ahead. kindly help on this.
ideviceinfo 1.3.0 is version i am using

nikias, 2024-06-13:
That's not a bug, the ProductType is not the model name.

closed 2024-06-13 by nikias

Lohithlee, 2024-06-13:
ProductType: iPhone12,1
above is the return for productType, could you please explain what it is provided that the connected phone was iPhone11

nikias, 2024-06-13:
https://theapplewiki.com/wiki/IPhone_11
see "InternalName"

https://ipsw.me/iPhone12,1/info
see "Identifier"

it's just how Apple named them internally...

mexmer, 2024-06-14:
as nikias said, it's internal product identification apple gave it, there are many sites, where you can search by that name (including apple support). you can also check this gist
https://gist.github.com/adamawolf/3048717
or if you interested also in apple computers and not just mobile devices (for example MacBookAir1,1 is returned for macbook air 13" from 2008) check this site
https://everymac.com/

Lohithlee, 2024-06-14:
> as nikias said, it's internal product identification apple gave it, there are many sites, where you can search by that name (including apple support). you can also check this gist [...]

thanks a lot