libimobiledevice/libimobiledevice #1572: ideviceinfo -q com.apple.disk_usage is showing wrong output of Storage value for iOS 17

evalamerr, 2024-06-19:
I have send this command to iOS 17 device & it is showing me the wrong output of storage value as compared to what is shown in the device.

Like I have 38.81 GB available in my device but this command shows AmountDataAvailable  is 30.7 GB & TotalDataAvailable  is 69.7 GB.

I used to use TotalDataAvailable  as the correct output of storage, but now TotalDataAvailable is much higher.  Is there something missing calculations??

Can someone help me in letting me know what is the change is iOS 17, why we are getting wrong values.??

mexmer, 2024-06-19:
TotalDataAvailable supposedly should be value of maximum storage available for user data (which also includes applications), excluding reserved space.
what it means in reality, and how it matches with what you see on device UI, you would need to ask apple engineers.
information provided change overtime, for example this stats you would get on iOS 3
```
WebAppCacheUsage : 0
PhotoUsage : 0
CalendarUsage : 110592
TotalDiskCapacity : 16239927296
CameraUsage : 0
AmountCameraAvailable : 15510867968
TotalDataAvailable : 15678640128
TotalDataCapacity : 15715639296
MobileApplicationUsage : 0
NotesUsage : 32768
AmountDataReserved : 167772160
MediaCacheUsage : 0
AmountDataAvailable : 15510867968
TotalSystemAvailable : 76472320
TotalSystemCapacity : 524288000
VoicemailUsage : 28672

```
and same device with iOS 4 (notice difference not only in ammount of information provided, but also even in total capacity)
```
TotalSystemAvailable : 76587008
AmountDataReserved : 167772160
TotalSystemCapacity : 713031680
MobileApplicationUsage : 154529792
TotalDataCapacity : 15527268352
TotalDiskCapacity : 16240300032
AmountDataAvailable : 14855102464
TotalDataAvailable : 15022874624
```

mind this value is queried directly from device OS, so whatever it reports, libimobile is not responsible for accuracy of this value.

evalamerr, 2024-06-19:
thanks for your answering and showing me the differences between each iOS version's usage info.  

However I don't know how to transfer this value to match what I see on device UI, **maybe I need the computational formula...**

For now, I just can use idevicesyslog to search system log and find the value of device UI when I occasionally tiggerered perference refresh( open General->About ). And this way is quiet inconvenience for me.
for example:
> Preferences(CacheDelete)[340] <Notice>: 659 publicClientAvailableSpaceForVolume kCacheDeleteAvailableSpaceClassEssential 38864684890 (38.86 GB) = freespace 30884282368 + purgeableSpace: 7980402522
> Preferences(CacheDelete)[340] <Notice>: CacheDeleteCopyAvailableSpaceForVolume result: 38864684890

mexmer, 2024-06-19:
honestly, i have no idea either, that's why i said, you would need apple engineer. for anyone else, it's just guesswork, unless you have information from their internal DK, this stuff is not documented anywhere.