libimobiledevice/libimobiledevice #1573: error when linking against so: `libimobiledevice`: missing symbols `userpref_get_paired_udids`, `userpref_read_pair_record` and `pair_record_get_host_id`

C0D3-M4513R, 2024-06-19:
### Steps To Reproduce
Steps to reproduce the behavior:
1. get the `libimobiledevice` so 
2. use `nm -gD` on the `libimobiledevice` so
3. don't find the listed functions

### Expected behavior
A clear and concise description of what you expected to happen.
The functions are there, and available to the linker.

### Additional context
Add any other context about the problem here.

I originally found this whilst trying to compile https://github.com/osy/Jitterbug/ on nixos - https://github.com/NixOS/nixpkgs/issues/320883.
In that issue the maintainer of the `libimobiledevice` package in nixos said:
> Looks like this might be an issue which should be reported upstream, it seems the linker isn't linking things from `common` as a symbol that is accessible.

I added some patches to Jitterbug whilst trying this, because there were some name conflicts between old source versions of `libimobiledevice-glue` and the `libplist` headers.
[0001-fix-some-stuff-on-nixos.patch.txt](https://github.com/user-attachments/files/15892764/0001-fix-some-stuff-on-nixos.patch.txt)
I used the following environment: `nix-shell -p ninja meson libgcrypt.dev libgcrypt libusbmuxd usbmuxd2 libimobiledevice-glue.dev libimobiledevice-glue libimobiledevice libunistring.dev pkg-config gnutls.dev`
(pretty much the same as in the Readme except usbmuxd2, which i bumped from usbmuxd)

<details><summary>Compile&Link of Jitterbug</summary>
<p>

```
[user@nixos-laptop:~/Documents/Jitterbug]$ nix-shell -p ninja meson libgcrypt.dev libgcrypt libusbmuxd usbmuxd2 libimobiledevice-glue.dev libimobiledevice-glue libimobiledevice libunistring.dev pkg-config gnutls.dev

[nix-shell:~/Documents/Jitterbug]$ rm -r build && meson build && cd build && meson compile
The Meson build system
Version: 1.4.0
Source dir: /home/user/Documents/Jitterbug
Build dir: /home/user/Documents/Jitterbug/build
Build type: native build
Project name: jitterbugpair
Project version: undefined
C compiler for the host machine: gcc (gcc 13.2.0 "gcc (GCC) 13.2.0")
C linker for the host machine: gcc ld.bfd 2.41
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: YES (/nix/store/zdvrzlvzbn9ymb0z8na50w995j8np16z-pkg-config-wrapper-0.29.2/bin/pkg-config) 0.29.2
libgcrypt-config found: YES (/nix/store/1jhhafn26f1wvq6n992qzc8kx4sb27cn-libgcrypt-1.10.3-dev/bin/libgcrypt-config) 1.10.3
Run-time dependency libgcrypt found: YES 1.10.3
WARNING: Static library 'usbmuxd-2.0' not found for dependency 'libusbmuxd-2.0', may not be statically linked
WARNING: Static library 'imobiledevice-glue-1.0' not found for dependency 'libusbmuxd-2.0', may not be statically linked
WARNING: Static library 'plist-2.0' not found for dependency 'libusbmuxd-2.0', may not be statically linked
Run-time dependency libusbmuxd-2.0 found: YES 2.0.2+date=2023-04-30
WARNING: Static library 'imobiledevice-1.0' not found for dependency 'libimobiledevice-1.0', may not be statically linked
WARNING: Static library 'usbmuxd-2.0' not found for dependency 'libimobiledevice-1.0', may not be statically linked
WARNING: Static library 'imobiledevice-glue-1.0' not found for dependency 'libimobiledevice-1.0', may not be statically linked
WARNING: Static library 'ssl' not found for dependency 'libimobiledevice-1.0', may not be statically linked
WARNING: Static library 'crypto' not found for dependency 'libimobiledevice-1.0', may not be statically linked
WARNING: Static library 'plist-2.0' not found for dependency 'libimobiledevice-1.0', may not be statically linked
Run-time dependency libimobiledevice-1.0 found: YES 1.3.0+date=2023-04-30
Build targets in project: 1

Found ninja-1.11.1 at /nix/store/hlxymn46n1xv6d0lq2jmcbl9b6g2l8s9-ninja-1.11.1/bin/ninja
WARNING: Running the setup command as `meson [options]` instead of `meson setup [options]` is ambiguous and deprecated.
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /nix/store/hlxymn46n1xv6d0lq2jmcbl9b6g2l8s9-ninja-1.11.1/bin/ninja
[1/2] Compiling C object jitterbugpair.p/JitterbugPair_main.c.o
In file included from /nix/store/fwh4fxd747m0py3ib3s5abamia9nrf90-glibc-2.39-52-dev/include/bits/libc-header-start.h:33,
                 from /nix/store/fwh4fxd747m0py3ib3s5abamia9nrf90-glibc-2.39-52-dev/include/stdio.h:28,
                 from ../JitterbugPair/main.c:17:
/nix/store/fwh4fxd747m0py3ib3s5abamia9nrf90-glibc-2.39-52-dev/include/features.h:414:4: warning: #warning _FORTIFY_SOURCE requires compiling with optimization (-O) [-Wcpp]
  414 | #  warning _FORTIFY_SOURCE requires compiling with optimization (-O)
      |    ^~~~~~~
../JitterbugPair/main.c: In function ‘main’:
../JitterbugPair/main.c:92:30: warning: passing argument 2 of ‘getopt’ from incompatible pointer type [-Wincompatible-pointer-types]
   92 |     while ((c = getopt(argc, argv, "lu:c")) != -1) {
      |                              ^~~~
      |                              |
      |                              const char **
In file included from /nix/store/fwh4fxd747m0py3ib3s5abamia9nrf90-glibc-2.39-52-dev/include/bits/getopt_posix.h:27,
                 from /nix/store/fwh4fxd747m0py3ib3s5abamia9nrf90-glibc-2.39-52-dev/include/unistd.h:903,
                 from ../JitterbugPair/main.c:20:
/nix/store/fwh4fxd747m0py3ib3s5abamia9nrf90-glibc-2.39-52-dev/include/bits/getopt_core.h:91:46: note: expected ‘char * const*’ but argument is of type ‘const char **’
   91 | extern int getopt (int ___argc, char *const *___argv, const char *__shortopts)
      |                                 ~~~~~~~~~~~~~^~~~~~~
[2/2] Linking target jitterbugpair
FAILED: jitterbugpair 
gcc  -o jitterbugpair jitterbugpair.p/JitterbugPair_main.c.o -Wl,--as-needed -Wl,--no-undefined -L/nix/store/6c947fh1y2ln5cr8kpcl14b3hfnni0av-libgcrypt-1.10.3/lib -Wl,--start-group -lgcrypt -L/nix/store/193fnfad15rgch01xxg7vf81rmmfwp8w-libgpg-error-1.48/lib -lgpg-error -L/nix/store/r45va2ilqijj85y8ig87lsh056cr452l-libusbmuxd-2.0.2+date=2023-04-30/lib -L/nix/store/8by7lhbpdjsy7r1b1cs7csfzz7xypdim-libplist-2.4.0/lib -L/nix/store/rm8hvv4l6jh8dlb44fy0s3snih9npflp-libimobiledevice-glue-1.2.0/lib -lusbmuxd-2.0 -limobiledevice-glue-1.0 -lplist-2.0 -ldl -pthread -L/nix/store/r45va2ilqijj85y8ig87lsh056cr452l-libusbmuxd-2.0.2+date=2023-04-30/lib -L/nix/store/77fr01d818vwavlpiakfq7166xmsnl64-libimobiledevice-1.3.0+date=2023-04-30/lib -L/nix/store/gp504m4dvw5k2pdx6pccf1km79fkcwgf-openssl-3.0.13/lib -L/nix/store/8by7lhbpdjsy7r1b1cs7csfzz7xypdim-libplist-2.4.0/lib -L/nix/store/rm8hvv4l6jh8dlb44fy0s3snih9npflp-libimobiledevice-glue-1.2.0/lib -limobiledevice-1.0 -lusbmuxd-2.0 -limobiledevice-glue-1.0 -lssl -lcrypto -lplist-2.0 -Wl,--end-group
/nix/store/bgcaxhhxswzvmxjbbgvvaximm5hwghz1-binutils-2.41/bin/ld: jitterbugpair.p/JitterbugPair_main.c.o: in function `print_udids':
/home/user/Documents/Jitterbug/build/../JitterbugPair/main.c:55:(.text+0x13d): undefined reference to `userpref_get_paired_udids'
/nix/store/bgcaxhhxswzvmxjbbgvvaximm5hwghz1-binutils-2.41/bin/ld: jitterbugpair.p/JitterbugPair_main.c.o: in function `main':
/home/user/Documents/Jitterbug/build/../JitterbugPair/main.c:162:(.text+0x623): undefined reference to `userpref_read_pair_record'
/nix/store/bgcaxhhxswzvmxjbbgvvaximm5hwghz1-binutils-2.41/bin/ld: /home/user/Documents/Jitterbug/build/../JitterbugPair/main.c:164:(.text+0x65b): undefined reference to `pair_record_get_host_id'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
</p>
</details>

I am on `x86_64-linux - Linux 6.6.31-rt31`

mexmer, 2024-06-19:
i advise you to compile everything from HEAD, seems you have mixed versions of libraries, that vere released at different dates, therefore there are inconsistencies. otherwise you would also not need your patch.

C0D3-M4513R, 2024-06-19:
> i advise you to compile everything from HEAD, seems you have mixed versions of libraries, that vere released at different dates, therefore there are inconsistencies. otherwise you would also not need your patch.

What is that going to change? All the function that fail to be found are there in the code. Even at the commit the .so was compiled from.