libimobiledevice/libimobiledevice #1590: idevice_event_subscribe callback not called after granting permission to photos and vidoes.

scottandrewzip, 2024-07-31:
I am using idevice_events_subscribe to listen for devices being connected and disconnected. This works 99% of the time however if you get a prompt on the phone to allow for photo and video access there is no callback telling you the the device is connected. If you disconnect and connect again you get the callback.

It seems with iOS 17.5.1 you get this message every 4th or 5th time you connect the device.

mexmer, 2024-07-31:
access to photos/videos is not cause by libimobile or usbmuxd, it is caused by MTP/WPD driver in your OS.
to access photos from computer, iOS/iPadOS has implemented Media Transfer Protocol (MTP) and it's extensions Windows Portable Device (WPD), this is universal protocol supported by most digital cameras, also by android phones, and many media devices (don't confuse it with mass storage, which is generaly used by external harddrives and usb sticks). 

MTP/WPD  works, even if you don't have libimobiledevice installed at all. and libimobile doens't know, if and when your computer access iOS device trough this protocol.

scottandrewzip, 2024-07-31:
Thats not the issue.. I don't want access.. The problem is the device callback is not called when that the device has attached. This dialog is stopping `idevice_events_subscribe` callback from being called. We do not get an event that the device was attached. I never said it was caused by it. Just that we don't know the device is reattached.

mexmer, 2024-07-31:
depending what your system configuration is, when you plug device to USB, it's handled either by systemd or udevd.

depending what your system configuration is, either you running usbmuxd as service, or is launched by udevd rule, or is launched by systemd rule.

until usbmuxd get information, that device is on usb it will not forward such infromation to libimobiledevice, therefore check your system log, if usbmuxd has been invoked, or usbmuxd log, if it got notification, that device is connected.

photos (MTP) and lockdown (which is used to inquire ios device information, and also to start services on ios devices) are on separate interface, and each is handled differently. 

so like i said, i would recommend checking what actually happens on ubsmuxd side (you can run it manually in foreground with higher verbosity). until usbmuxd does preflight, it will not report device further along the path.