libimobiledevice/libimobiledevice #1596: error Building libimobiledevice-glue (configure: error: pkg-config not found)

Lautarob1, 2024-08-17:
Using a Libimobile in a Mac Intel, I constantly get error Started "com.apple.mobilebackup2" service on port 54332.
Could not perform backup protocol version exchange, error code -1
Since I installed libimobiledevice using Brew, and I read somewhere here that this could originate the problem (brew version not updated recently). I tried to reinstall libimobile using the script limb-build-macos.sh (version 1.0.17). 

It worked for a while, doing a bunch of things, but right after requesting the admin password, it breaks and the log file has the following error at the very end (last 2 lines)
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for pkg-config... no
configure: error: pkg-config not found

It seems to be an easy problem, but I have no idea about which pkg-config is missing and how to get it and place in the appropriate folder...

mexmer, 2024-08-19:
you can try build script made by nikias
https://gist.github.com/nikias/84c79469a1d0f16ff95250f0d51858c3

Lautarob1, 2024-08-19:
Thank you! mexmer... That is precisely what I was attempting to to when got the error mentioned ("_I tried to reinstall libimobile using the script limb-build-macos.sh (version 1.0.17)._" I guess this might be a simple error, but I couldn't find a solution: no idea which package is missing and where it may be or should be located...

mexmer, 2024-08-20:
sorry, missed that part,
_brew install pkg-config_ 
or 
_brew install pkgconfig_ 
(i think later one is alias, because it was without - at some point of time)
pkg-config is normal tool that most developers use, when working with custom libraries, so i can imagine it's not mentioned in prerequisities, same like m4 or autoconf, just standard tools you use.

Lautarob1, 2024-08-20:
Thank you Sir!. I will try ASAP an let you know the outcome.

Lautarob1, 2024-08-21:
Now I got a different error: ERROR: configure failed for libimobiledevice
(referred to the file /limd-build/libimobiledevice_configure.log
The error refers to (looking at the final lines of :/libimobiledevice_configure.log)

Configuration for libimobiledevice-glue 1.3.0:
-------------------------------------------

  Install prefix: .........: /usr/local

  Now type 'make' to build libimobiledevice-glue 1.3.0,
  and then 'make install' for installation.
Configuration for libimobiledevice-glue 1.3.0:
-------------------------------------------

Surely this is something easy to understand/implement for an experienced developer, but that is not my case, even less in MacOS.  Can you please guide me?

Where the "install prefix" should be used and how? 
Where the "make" command should be used and in what context?  (make what?)
How shoud I use "make install" (which folder on the terminal I am supposed to be in?)

Any help will be highly appreciated,

Lautaro

mexmer, 2024-08-21:
i would recommend  either running limd script i linked before in different folder, or clear the one you have sources now. autoconf caches results to speedup repeated runs, which might sometime result in unexpected behavior.

on other note, sorry to say that, but if you don't even know, how **make** works and what is it good for, then you should leave this to someone else. **make** is one of basic commandline developer tools, that runs other tools, according to prescription from Makefile