libimobiledevice/libimobiledevice #1597: Can the homescreen.png and lockscreen.png be captured using libimobileDevice?

Lautarob1, 2024-08-18:
There are a few apps that as soon get connected to the iOS device, the homescreen.png and lockscreen.png are immediately captured and showed in the app likewise the serial number and other device info is (or can be) captured. Thus, you can have the said homescreen.png available exactly as you have the device info or ID (immediately it get connected).

My question is if the said capture of homescreen.png can be implemented using libimobiledevice framework, If so, how?. If not, why not?

Thanks!!

mexmer, 2024-08-19:
screencapture is done with screenshot service, this is supported up to ios version 16, for version 17 and newer it's not implemented yet.

you might want to look at https://github.com/doronz88/pymobiledevice3 which has implemented support for ios17 development tools, and i think screenshot is working in that project.

Lautarob1, 2024-08-20:
Thank you mexmer, but although surely while a screenshot feature well implemented could solve the problem, I was referring to something different. 

As I said, there are a few apps that as soon as the iPhone get connected it grab the homescreen and shows it on the app (kind of automatically). Probably these app can take screenshots as well, but that is a different story. They do exactly the same when an iPad is connected. 

There must be some low level command (like, just as example, ideviceinfo) that permit to grab the homescreen (and also, at the same time, the lockscreen). Furthermore, when an iPhone is connected and the said apps are running, the home screen is continuously seen even when it changes (for example, for those iPhones that have a bunch of pictures that varies constantly).

Again, thanks for taking the time to help me understand this issue,

All the best,'

Lautaro

mexmer, 2024-08-20:
give me example of such application.
only application i use for this type of interaction are developer tools, and they either utilize developer image (which is not supported for iOS 17 and newer), automationui framework or webdriver (which both also requires some setup).

mexmer, 2024-08-20:
trough springboard service you can retrieve wallpaper and icons position, so maybe there are other options, that can be used, that can get lockscreen. only thing that comes to my mind, that doesn't require driver o developer tools disk.

Lautarob1, 2024-08-20:
Thanks mexmer. I'll appreciate that you take the time to review and answer my question. If any additional ideas on how this is done comes to you your mind, please share with me.
On the other hand, can you give an idea on how to use springboard from the MacOS terminal, if that makes sense?

mexmer, 2024-08-20:
not sure what you mean by that. some functionality of springboard service is implemented and interface is exposed, check this header https://github.com/libimobiledevice/libimobiledevice/blob/master/include/libimobiledevice/sbservices.h

Lautarob1, 2024-08-21:
Thanks mexmer, 

Just for the record, I posted a similar question on the Apple Dev forum and got this answer:

- _There’s no supported API for that.
- Some folks have reverse engineered Apple’s internal mechanisms for this._

So,  it looks like no immediate solution is available. If you have an general idea on how to address this task, please let me know. 

Thanks again for your help, I very much appreciate it.

Lautaro