libimobiledevice/libimobiledevice #1598: Installation Failed, could not verify app integrity.

robert-hopla, 2024-08-20:
hello, we have an app to test phone hardware. It usually is installed on phones without icloud account, hence the appstore way is very user unfriendly. 

we have a setup with Volume Purchasing, that works, but requires a verified ABM account, which is another hurdle and delay.

we stumbled upon ilibmobile device, and it looks very promising (!), but we are not able to launch the app. What are we missing, and how to fix the below errors?

[image]
[image]

robert-hopla, 2024-08-20:
our goal is to create a small desktop app that installs our app on devices connected to the computer via USB...

and, eventually, factory reset, skip setup assistant, and install app..

juliohopla, 2024-08-20:
@mexmer

mexmer, 2024-08-21:
i think error on your console is selfexplaining, your application package is missing provisioning profile, or is not properly signed.
unless device is jailbroken, sideloading is only allowed either for applications with developer provisioning profile (which requires device registered as development device), or when application is signed with enterprise certificate, but for that you need install proper provisioning profile (this is what alt stores use to install apps)

you can also temporary sign app with user apple id (i think signature is valid only for 7 days, after that application willl stop working), but i have never used that, so i have no idea how it's working. i think Appium use this, when you don't have developer account.

robert-hopla, 2024-08-21:
thanks @mexmer do you have more information on how to get the temporary provisioning profile, 7 days is enough for our usecase.

enterprise we don't qualify for.

mexmer, 2024-08-21:
like i said, no idea, never used that.