libimobiledevice/libimobiledevice #1606: ERROR: Could not connect to lockdownd: SSL error (-5)

SoloRobo, 2024-09-24:
On Fedora 41 Beta using libmobiledevice - 1.3.0^20230705git6fc41f5 and libimobiledevice-utils - 1.3.0^20230705git6fc41f5 after trustung the iDevice I get:

```
ideviceinfo 
ERROR: Could not connect to lockdownd: SSL error (-5)
```

This is using an iPhone SE 15.8.2 I have an iPad Air 2 15.8.2 exhibiting the same issue.

SoloRobo, 2024-09-24:
Experiencing the same error on iPad Pro 10.5 17.7

mexmer, 2024-09-24:
ssl error is most likely sign of expired pairing record. do 
`idevicepair unpair`
then 
`idevicepair pair`
have only connected one device at time, othwerise you will also need to specify udid

mexmer, 2024-09-24:
if it's not paired, then pair itself shoudl work, so if you confirm trust, ideviceinfo still doesn't work?

SoloRobo, 2024-09-24:
Correct, it doesn't work after I click "Trust", it never pairs.

I've tried on iDevices I've used before and an iPad I hadn't used with the same result.

I wonder, is there a cache or set of files I can manually remove, perhaps the Fedora 40 to Fedora 41 Beta upgrade has broken something?

mexmer, 2024-09-24:
you might need to build from source then, to have latest version. i thought only some tools were affected and version 1.3.0 should be fine, not ideviceinfo, but seems it needs update too.

kohnish, 2024-09-25:
Linking with openssl built with enable-weak-ssl-ciphers fixes the issue.

mexmer, 2024-09-26:
> Linking with openssl built with enable-weak-ssl-ciphers fixes the issue.

i didn't expect ios 15 to use ciphers from weak cipher list, but you are right. those defaults in builds change every so offten.

kohnish, 2024-09-26:
I feel like this project needs to somehow address the issue. It's unlikely binary distros will keep providing openssl with weak cipher support.  
Maybe release a static binary for major architectures in github.  
It is a massive pain to build usbmuxd with all these micro repositories.

raaffaaeell, 2024-09-27:
Sorry if I'm cluttering the thread, but I updated my system to Fedora 41 too and I cant pair my iphone either. Getting the same SSL error, but I have my phone (iphone 13) on iOS 18 already

kohnish, 2024-09-28:
> Sorry if I'm cluttering the thread, but I updated my system to Fedora 41 too and I cant pair my iphone either. Getting the same SSL error, but I have my phone (iphone 13) on iOS 18 already
  
If you are on x86_64 fedora 41, you can replace your /usr/sbin/usbmuxd with what's in this zip file if you trust me to see if it's cause by openssl from fedora 41.  

[usbmuxd-weak-cipher.zip](https://github.com/user-attachments/files/17173785/usbmuxd-weak-cipher.zip)

SoloRobo, 2024-09-29:
I've replaced usbmuxd with usbmuxd-weak-cipher as requested and can confirm that I can now open Documents on my iDevice.

mexmer, 2024-09-30:
> If you are on x86_64 fedora 41, you can replace your /usr/sbin/usbmuxd with what's in this zip file if you trust me to see if it's cause by openssl from fedora 41. [...]

might be good if you submit PR, if there is need to change something in usbmuxd code,

kohnish, 2024-09-30:
>  if there is need to change something in usbmuxd code,  
   
No code changes required.

SoloRobo, 2024-10-03:
Would I be right in thinking there is a corner of Fedora-land where we need to ask nicely to compile usbmuxd differently?

kohnish, 2024-10-03:
> Would I be right in thinking there is a corner of Fedora-land where we need to ask nicely to compile usbmuxd differently?
  
I asked the maintainer of libimobiledevice package to link with gnutls which is also another workaround.

SoloRobo, 2024-10-03:
That's great, I guess my concern is the build Fedora appear to be using to my amateur eye is a build from commit: https://github.com/libimobiledevice/libimobiledevice/commit/6fc41f57fc607df9b07446ca45bdf754225c9bd9

Which relates, I assume to the version number get from 'dnf info libimobliedevice': 1.3.0^20230705git6fc41f5

SoloRobo, 2024-10-31:
Update on this, there is now a bug opened with Fedora/RedHat here:

https://bugzilla.redhat.com/show_bug.cgi?id=2321958

```
sudo update-crypto-policies --set FEDORA40
```

The above solves it in the very short term and the bug report above has more detail (I barely understand)

acheong08, 2024-11-08:
@SoloRobo Thank you. Interestingly enough, I have the same problem on Arch. Not sure how to solve it there but at least it works on my fedora machine now.

nikias, 2024-11-09:
libimobiledevice can also be linked against mbedtls. That's licensed under Apache 2.0 or GPL 2.0 so it would not even be a problem to distribute it with the project.
In all cases, SHA-1 being deprecated from public up-to-date packages is outside of our control, in that case a different "insecure" package would have to be provided by the distributions, causing even more chaos I guess. On macOS Apple ships their own LibreSSL build which to my knowledge still has support for it.

kohnish, 2024-11-10:
I was looking into @SoloRobo 's workaround which suggests runtime workaround is possible. 
 
It turns out Fedora has this very aggressive patch. So you can set the custom environment variable for the process as well. 

https://src.fedoraproject.org/rpms/openssl/blob/rawhide/f/0049-Allow-disabling-of-SHA1-signatures.patch  
  
I agree that the project shouldn't bother supporting cases like this unless it's merged to the real upstream. Sha1 usage for root cert is quite common as its strength doesn't matter.   According to my observation, latest vanilla openssl is already supported by the project.

lolcatw, 2024-11-22:
How could one re-compile usbmuxd to support weak ciphers? I'm on debian, and I'm trying to pair a ipod touch 4g running ios 6.1.6

kohnish, 2024-11-22:
> How could one re-compile usbmuxd to support weak ciphers? I'm on debian, and I'm trying to pair a ipod touch 4g running ios 6.1.6
  
By linking with such openssl build.   
But the openssl weak-ciphers thing I mentioned was mistaken, probably vanilla openssl also works.  
The issue is fedora specific where signing with sha1 causes error by the patch I linked.  
Every other distribution should be fine.

hmlnarik, 2024-12-14:
FWIW, as a single-shot workaround, one can use the weakened OpenSSL configuration instead of lowering the whole system security standards. This worked for me to pair the device, and no other trick was needed after pairing to work with the files in the Apple device.

1) Store the following OpenSSL configuration into a file, e.g. `$HOME/openssl-weak.conf`
```
.include /etc/ssl/openssl.cnf
[openssl_init]
alg_section = evp_properties
[evp_properties]
rh-allow-sha1-signatures = yes
```
2) Run desired command with the altered OpenSSL configuration, e.g.:
`$ OPENSSL_CONF=$HOME/openssl-weak.conf idevicepair pair <device-id>`

Inspired by https://community.n8n.io/t/ssh-node-to-legacy-systems/57703

mexmer, 2024-12-16:
i'm not sure if this will work since it's usbmuxd and not idevicepair, that is generating pairing certificates. but it's neat idea.

hmlnarik, 2024-12-16:
> i'm not sure if this will work since it's usbmuxd and not idevicepair, that is generating pairing certificates. but it's neat idea.

Works for me for pairing the device, just retried again (twice). Browsing in Plasma's Dolphin and GNOME's Files fails with aforementioned error message upon connecting. Once paired [as outlined above](https://github.com/libimobiledevice/libimobiledevice/issues/1606#issuecomment-2543116644), I can browse the device in Plasma's Dolphin as well as in GNOME's Files.

This would indicate it may be not an issue with usbmuxd but rather with pairing as driven by idevicepair.

mexmer, 2024-12-16:
i just checked and certifcate generating function is inside libimobiledevice ... usbmuxd only stores them to lockdown folder
i though usbmuxd creates certificates during preflight for unknown devices, since that is not the case, your solution should work. 

so yes, that  should save headache for lot of people.

bipox, 2025-04-14:
To get the device ID required for [this fix](https://github.com/libimobiledevice/libimobiledevice/issues/1606#issuecomment-2543116644):

`lsusb -v 2> /dev/null | grep -e "Apple Inc" -A 2`

dustymabe, 2025-04-14:
> To get the device ID required for [this fix](https://github.com/libimobiledevice/libimobiledevice/issues/1606#issuecomment-2543116644):
> 
> `lsusb -v 2> /dev/null | grep -e "Apple Inc" -A 2`

YMMV but for me if you just leave off the `<device-id>` the command finds the device attached to my system (of course I only have one iphone attached).

avibrazil, 2025-10-14:
With the OpenSSL tricks provided here, everything works.

With USB cable only.

Anyone capable of pairing and backing up through WiFi?

I'm on Fedora 42, iOS 26, usbmuxd-1.1.1^20240915git0b1b233-4.fc42, libimobiledevice-1.3.0^20240916gited9703d-5.fc42, libimobiledevice-utils-1.3.0^20240916gited9703d-5.fc42

What is required to connect and backup via WiFi on Fedora 42 as a regular Linux user ?

velosol, 2025-10-17:
@avibrazil The [docs](https://github.com/libimobiledevice/libimobiledevice/blob/master/docs/idevicepair.1) say wireless pairing is only available with Apple TV devices as of release [1.4.0 10-Oct-2025](https://github.com/libimobiledevice/libimobiledevice/releases/tag/1.4.0):

```
Pairing over network (wireless pairing) is only supported by Apple TV
devices. To perform a wireless pairing, you need to use the -w
command line switch.
```

Further down that note is more bad news: 
`Please note that wireless pairing is currently not supported on Linux.`

mexmer, 2025-10-17:
while pairing is not supported, you can pair over cable, then switch sync to wifi, that's something that iOS devices support for long time.

absence of pairing over network is not problem inside libimobile, it's how apple made it, Apple TV is exception, because there is no usb port in that one anymore. old Apple TV models had it, and you could use it to debug applications in Apple TV connected over USB, they removed it in version 3 or 4 of Apple TV.

tbh. no idea how vision pro is pairing to xcode, if it has usb cable or over wifi, since i don't own it.