libimobiledevice/libimobiledevice #1607: mobilegestalt PasswordConfigured return MobileGestaltDeprecated on different ios versions

irainbw, 2024-09-27:
idevicediagnostics  mobilegestalt PasswordConfigured
`<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>MobileGestalt</key>
        <dict>
                <key>Status</key>
                <string>MobileGestaltDeprecated</string>
        </dict>
</dict>
</plist>`

iphone15 pro   
iosVersion: 18b1

Another device (XR, iosVersion 15) behaves normally
How do I get whether to set a password on a higher version?

irainbw, 2024-09-27:
> idevicediagnostics mobilegestalt PasswordConfigured iphone15 pro iosVersion: 18b1
> 
> Another device (XR, iosVersion 15) behaves normally How do I get whether to set a password on a higher version?

[log.txt](https://github.com/user-attachments/files/17161355/log.txt)

zytzjx, 2025-10-24:
Apple remove mobilegestalt  supported.