libimobiledevice/libimobiledevice #1609: Build fails with modern compilers due to 25-year-old illegal constructs

eli-schwartz, 2024-10-07:
https://wiki.gentoo.org/wiki/Modern_C_porting#How_do_I_reproduce_these_bugs.3F

GCC 14 and clang 16 have started to error on invalid C code that was removed from the standard in c99 or even earlier. Historically, compilers accepted this code even though it was invalid in an effort to avoid breaking really old codebases. Due to the passing of time and an increasing understanding of just how dangerous this invalid code is, that leniency is going away.

Note that this is ***not*** cosmetic. It is undefined behavior and compilers will (increasingly) optimize it in weird and wonderful ways that break the original intention of the code. It will also obviously mean that the code cannot be built on newer systems with updated compilers.

An easy way to reproduce the problem even on older compiler versions is to build with the following *FLAGS:

```
-Werror=implicit-function-declaration -Werror=implicit-int -Werror=int-conversion -Werror=incompatible-pointer-types
```

libimobiledevice fails with the following error:
```
libtool: compile:  x86_64-pc-linux-gnu-gcc -DHAVE_CONFIG_H -I. -I../.. -I../../include -I../.. -Wno-incompatible-pointer-types -DHAVE_CONFIG_H -DOPENSSL=1 -pipe -march=native -fstack-protector-all -O2 -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wformat -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=int-conversion -Werror=incompatible-pointer-types -c cstr.c  -fPIC -DPIC -o .libs/cstr.o
cstr.c:24:44: error: initialization of 'void * (*)(size_t,  void *)' {aka 'void * (*)(long unsigned int,  void *)'} from incompatible pointer type 'void * (*)(size_t)' {aka 'void * (*)(long unsigned int)'} [-Werror=incompatible-pointer-types]
   24 | static cstr_allocator malloc_allocator = { malloc, free, NULL };
      |                                            ^~~~~~
cstr.c:24:44: note: (near initialization for 'malloc_allocator.alloc')
cstr.c:24:52: error: initialization of 'void (*)(void *, void *)' from incompatible pointer type 'void (*)(void *)' [-Werror=incompatible-pointer-types]
   24 | static cstr_allocator malloc_allocator = { malloc, free, NULL };
      |                                                    ^~~~
cstr.c:24:52: note: (near initialization for 'malloc_allocator.free')
cc1: some warnings being treated as errors
make[3]: *** [Makefile:435: cstr.lo] Error 1
make[3]: Leaving directory '/var/tmp/portage/app-pda/libimobiledevice-9999/work/libimobiledevice-9999/3rd_party/libsrp6a-sha512'
```

There is some questionable code in this file:
https://github.com/libimobiledevice/libimobiledevice/blob/ed9703db1ee6d54e3801b618cee9524563d709e1/3rd_party/libsrp6a-sha512/cstr.c#L13-L25

It is not actually okay, but that's beside the point. Even setting this magic undocumented define which is supposed to "make the code handle the case where it is totally wrong to violate the C standard", it still doesn't actually work. :)
```
libtool: compile:  x86_64-pc-linux-gnu-gcc -DHAVE_CONFIG_H -I. -I../.. -I../../include -I../.. -Wno-incompatible-pointer-types -DHAVE_CONFIG_H -DOPENSSL=1 -pipe -march=native -fstack-protector-all -O2 -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wformat -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=int-conversion -Werror=incompatible-pointer-types -DPEDANTIC_ARGS -c cstr.c  -fPIC -DPIC -o .libs/cstr.o
cstr.c:22:44: error: initialization of 'void * (*)(size_t,  void *)' {aka 'void * (*)(long unsigned int,  void *)'} from incompatible pointer type 'void * (*)(int,  void *)' [-Werror=incompatible-pointer-types]
   22 | static cstr_allocator malloc_allocator = { Cmalloc, Cfree, NULL };
      |                                            ^~~~~~~
cstr.c:22:44: note: (near initialization for 'malloc_allocator.alloc')
cc1: some warnings being treated as errors
make[3]: *** [Makefile:435: cstr.lo] Error 1
make[3]: Leaving directory '/var/tmp/portage/app-pda/libimobiledevice-9999/work/libimobiledevice-9999/3rd_party/libsrp6a-sha512'
```

The libsrp6a library honestly looks kind of bad -- and basically abandonware? maybe worth rewriting from scratch to actually be conformant C.

nikias, 2024-10-08:
I just replaced the code in question with good old malloc and free: 881102944edf36abc98539b10d13e2375fda88cf

eli-schwartz, 2024-10-08:
Thanks! Much appreciated. After that commit, the build succeeds.

```
>>> Unpacking source...
 * Repository id: libimobiledevice_libimobiledevice.git
 * To override fetched repository properties, use:
 *   EGIT_OVERRIDE_REPO_LIBIMOBILEDEVICE_LIBIMOBILEDEVICE
 *   EGIT_OVERRIDE_BRANCH_LIBIMOBILEDEVICE_LIBIMOBILEDEVICE
 *   EGIT_OVERRIDE_COMMIT_LIBIMOBILEDEVICE_LIBIMOBILEDEVICE
 *   EGIT_OVERRIDE_COMMIT_DATE_LIBIMOBILEDEVICE_LIBIMOBILEDEVICE
 * 
 * Fetching https://github.com/libimobiledevice/libimobiledevice ...
git fetch https://github.com/libimobiledevice/libimobiledevice +HEAD:refs/git-r3/HEAD
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Total 6 (delta 5), reused 6 (delta 5), pack-reused 0 (from 0)
Unpacking objects: 100% (6/6), 695 bytes | 115.00 KiB/s, done.
From https://github.com/libimobiledevice/libimobiledevice
   ed9703d..8811029  HEAD       -> refs/git-r3/HEAD
git symbolic-ref refs/git-r3/app-pda/libimobiledevice/0/__main__ refs/git-r3/HEAD
 * Checking out https://github.com/libimobiledevice/libimobiledevice to /var/tmp/portage/app-pda/libimobiledevice-9999/work/libimobiledevice-9999 ...
git checkout --quiet refs/git-r3/HEAD
GIT update -->
   repository:               https://github.com/libimobiledevice/libimobiledevice
   updating from commit:     ed9703db1ee6d54e3801b618cee9524563d709e1
   to commit:                881102944edf36abc98539b10d13e2375fda88cf
 3rd_party/libsrp6a-sha512/cstr.c | 62 ++++++++++----------------------------------------------------
 3rd_party/libsrp6a-sha512/cstr.h | 12 ------------
 2 files changed, 10 insertions(+), 64 deletions(-)
>>> Source unpacked in /var/tmp/portage/app-pda/libimobiledevice-9999/work

[...]

/bin/sh ../libtool  --tag=CC   --mode=link x86_64-pc-linux-gnu-gcc -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden    -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -pipe -march=native -fstack-protector-all -O2 -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=int-conversion -Werror=incompatible-pointer-types -lplist-2.0 -lplist -lreadline -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wl,--defsym=__gentoo_check_ldflags__=0 -o afcclient afcclient-afcclient.o ../src/libimobiledevice-1.0.la -limobiledevice-glue-1.0 -lplist-2.0 -lplist -lpthread 
libtool: link: x86_64-pc-linux-gnu-gcc -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -pipe -march=native -fstack-protector-all -O2 -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wformat -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=int-conversion -Werror=incompatible-pointer-types -Wl,-O1 -Wl,-z -Wl,pack-relative-relocs -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wl,--defsym=__gentoo_check_ldflags__=0 -o .libs/afcclient afcclient-afcclient.o  -lreadline -Wl,--as-needed ../src/.libs/libimobiledevice-1.0.so -lssl -lcrypto -lusbmuxd-2.0 -limobiledevice-glue-1.0 -lplist-2.0 -lplist -lpthread -pthread
make[2]: Leaving directory '/var/tmp/portage/app-pda/libimobiledevice-9999/work/libimobiledevice-9999/tools'
Making all in docs
make[2]: Nothing to be done for 'all'.
>>> Source compiled.
```

At some point it would be fantastic to have a new stable release as the 1.3.0 tag will be unable to build successfully on any distro that has updated to gcc 14. It is e.g. the default in Debian Trixie (currently the testing version of debian).

closed 2024-10-18 by nikias