libimobiledevice/libimobiledevice #1617: Static compilation is broken for WIN32

MarSoft, 2024-11-04:
I'm building `libimobiledevice` suite in static mode. I've cloned `libplist`, `libimobiledevice-glue`, `libusbmuxd` and `libimobiledevice` repositories, configured each of them with `--enable-shared=no` flag and am trying to build them.
But each depending project expects its dependency to be a shared library rather than static one. E.g. when I bulid `libusbmuxd` after installing static `libimobiledevice-glue`, I get errors related to `__imp_

Steps to reproduce:
1. Clone `libplist` and `libimobiledevice-glue` and `libusbmuxd`
2. Build `libplist` using `mingw`
```
$ PKG_CONFIG_LIBDIR=/tmp/test1/lib/pkgconfig ./autogen.sh --enable-shared=no --prefix=/tmp/test1 --host=x86_64-w64-mingw32
$ make && make install
```
3. Try building `libimobiledevice-glue` with the same commands. It will complain:
```
/usr/bin/x86_64-w64-mingw32-ld: ../src/.libs/libusbmuxd-2.0.a(libusbmuxd.o): in function `send_plist_packet':
/tmp/libusbmuxd/src/libusbmuxd.c:575:(.text+0x35bc): undefined reference to `__imp_plist_to_xml'
/usr/bin/x86_64-w64-mingw32-ld: ../src/.libs/libusbmuxd-2.0.a(libusbmuxd.o): in function `send_pair_record_packet':
/tmp/libusbmuxd/src/libusbmuxd.c:828:(.text+0x35ed): undefined reference to `__imp_plist_free'
```

This happens because of this part: https://github.com/libimobiledevice/libplist/blob/master/include/plist/plist.h#L78-L86
When we compiling `libimobiledevice-glue`, we don't have `LIBPLIST_STATIC` declared anymore (because it was only declared when building `libplist` but not exported anywhere).
I think we should somehow add `-DLIBPLIST_STATIC` to `Cflags` line in `libplist-2.0.pc`. Not sure how do we do this conditionally.

mexmer, 2024-11-04:
actually, when you compile whole libimobile set as static, you might want all of it static (so you can then have just 1 library)
in that case you would need
-DLIBPLIST_STATIC -DLIMD_GLUE_STATIC -DLIBUSBMUXD_STATIC -DLIBIMOBILEDEVICE_STATIC
i would not recommend in general mixing static and dynamic builds for all these, so either do all static, or all dynamic, but regardless, there comes 2 ways how to approach this in my mind

either when enabling static build for libimobile, add what i wrote above (and that implies all other projects are build as static) or use approach that i have seen on many other projects (like openssl, qt, and others) where you add more switches to configure usage of either dynamic or static 3rd party library.
assuming all builds will be by default dynamic, something like this
_--with-plist-static 
--with-libimobiledevice-glue-static
--with-libusbmuxd-static_
which will cause additional flag for respective library

tbh. since i compile libimobile with msvc, i just use Directory.Build.Props for preprocessor definitions to avoid too much modifications of vcxproj, but that's just one build system, therefore no general solution.

MarSoft, 2024-11-05:
Yes, I do want a fully static build. But the new approach with `smth_STATIC` constants implies that one should provide separate static flags for each library.
Since we already know if certain library is built in static-only mode, what's wrong with adding corresponding `-D` flag to its `pkgconfig` file?
Yes, `--with-smth-static` would be useful if corresponding library has both static and shared versions.

juanpenia, 2024-12-10:
i'm having the same issue but only when building libusbmuxd, i can successfully compile libimobiledevice-glue using this:

`./autogen.sh --enable-static --disable-shared`